openwrt/packages #20342: openvpn-openssl: requires compile option OPENVPN_openssl_ENABLE_IPROUTE2

alexw65500, 2023-01-18:
Maintainer: Magnus Kroken <mkroken@gmail.com>

Environment: ath79, WZR-HP-AG300H, v22.03.3

Description:
openvpn-openssl is unable to install routes pushed by the server when running as client on an openwrt since some time up to at least openwrt 22.03.3.

Here the relevant logs when openvpn tried to set the pushed routes with the "official" packet:

> Wed Jan 18 13:35:45 2023 daemon.notice openvpn(sample_client)[5534]: Outgoing Data Channel: Cipher 'AES-256-GCM' initialized with 256 bit key
Wed Jan 18 13:35:45 2023 daemon.notice openvpn(sample_client)[5534]: Incoming Data Channel: Cipher 'AES-256-GCM' initialized with 256 bit key
Wed Jan 18 13:35:45 2023 daemon.notice openvpn(sample_client)[5534]: net_route_v4_best_gw query: dst 0.0.0.0
Wed Jan 18 13:35:45 2023 daemon.notice openvpn(sample_client)[5534]: net_route_v4_best_gw result: via 192.168.0.1 dev eth2
Wed Jan 18 13:35:45 2023 daemon.notice openvpn(sample_client)[5534]: TUN/TAP device tun0 opened
Wed Jan 18 13:35:45 2023 daemon.notice openvpn(sample_client)[5534]: net_iface_mtu_set: mtu 1500 for tun0
Wed Jan 18 13:35:45 2023 daemon.notice openvpn(sample_client)[5534]: net_iface_up: set tun0 up
Wed Jan 18 13:35:45 2023 daemon.notice openvpn(sample_client)[5534]: net_addr_ptp_v4_add: 10.16.4.2 peer 10.16.4.1 dev tun0
Wed Jan 18 13:35:45 2023 daemon.notice openvpn(sample_client)[5534]: /usr/libexec/openvpn-hotplug up sample_client tun0 1500 1624 10.16.4.2 10.16.4.1 
init
Wed Jan 18 13:35:46 2023 daemon.notice openvpn(sample_client)[5534]: net_route_v4_add: 10.16.4.1/32 via 10.16.4.1 dev [NULL] table 0 metric -1
Wed Jan 18 13:35:46 2023 daemon.notice openvpn(sample_client)[5534]: net_route_v4_add: 10.16.1.0/24 via 10.16.4.1 dev [NULL] table 0 metric -1
Wed Jan 18 13:35:46 2023 daemon.warn openvpn(sample_client)[5534]: sitnl_send: rtnl: generic error (-128): Network unreachable
Wed Jan 18 13:35:46 2023 daemon.warn openvpn(sample_client)[5534]: ERROR: Linux route add command failed

setting the routes manually with e.g.
`ip route add 10.16.1.0/24 dev tun0`
restores the openvpn service.

As a more permanent solution I recompiled openvpn-openssl with "OPENVPN_openssl_ENABLE_IPROUTE2", which also fixes the issue. 

I suggest setting that parameter by default or - assuming all supported versions are now able to use iproute2 - drop that as option and always have it enabled.

Neustradamus, 2023-05-18:
@dhewg, @AuthorReflex, @neheb, @sch-m, @ynezz, @champtar, @PolynomialDivision, @mkrkn, @jow-, @egorenar, @miska: Have you seen this ticket?

neheb, 2023-05-28:
@cotequeiroz might know