openwrt/packages #20348: Why disable update checks in AdGuard Home?

CodeSapiens, 2023-01-19:
I have been searching all over the web for a way to enable the "Check for updates" button, and allow AdGuard Home to automatically check for updates. Turns out you guys are disabling that by adding the parameter --no-check-update to the install script (see here: https://github.com/openwrt/packages/blob/master/net/adguardhome/files/adguardhome.init).

I would like to know how to enable update checking without having to reinstall or compile a separate installer. Please consider removing that parameter from the install script.
The folks at GL.iNet don't do that with their OpenWRT package for AdGuard Home, which works great and without issues.

From what I gathered in my search, several people appear to be looking for this too, though I found no open issues in this forum.

Thanks!

brada4, 2023-01-20:
Because you used package manager to install binary package?
If you use binary from their repo (and it runs on musl i.e. does not use glibc specifics), it will update, subject to its ability to handle trimmed-down sysvinit support for process restart.

CodeSapiens, 2023-01-20:
> Because you used package manager to install binary package? If you use binary from their repo (and it runs on musl i.e. does not use glibc specifics), it will update, subject to its ability to handle trimmed-down sysvinit support for process restart.

Yes, I did use the package manager to install, as do thousands of people. That still does not answer the question of why the package providers disable update checks on the AGH package, when other OpenWRT providers do not. I am certainly not the only one frustrated by this, from what I have read on various forums.

brada4, 2023-01-20:
Because builtin update checks are against adguardhome binary repository, to update 40MB binary package installed from there, not suitable to update opkg, rpm, deb or any other packaging system installed files.

You could ping last commiter to get that +0.0.1 version via opkg @solomonricky @BKPepe

brada4, 2023-01-20:
> The folks who dont disable update check say

https://forum.gl-inet.com/t/adguardhome-on-gl-routers/10664/5

CodeSapiens, 2023-01-20:
> > The folks who dont disable update check say
> 
> https://forum.gl-inet.com/t/adguardhome-on-gl-routers/10664/5

After reading that blog I didn't get the impression that anything said presented enabling update checks as a problem. One person asked if the new release shown by the update check was handled by firmware updates, and the answer was simply that it was an official update from AGH, not from GL.iNet. I saw no complaints about that.

What I can say from my own experience as a GL.iNet router owner is, that installing from their package manager only installs the version they included in their initial package and does not immediately try to download an additional 40MB binary, as I think you suggested. Once the installation is completed and you launch the UI, then and only then do you see the prompt to install an available update, which is wonderful, and totally optional to the user. Also, the "Check for updates now" button is visible at the bottom of the page as it should, which is not available in this other AGH installation, due to the handicapped installation.

Thanks for your time and for following up on this thread Brada4. I still believe this should be given serious consideration, and leave the option to check for updates and install updates when available, up to the user. Having to wait for package deployments via LuCI can sometimes take a long time, depending on how frequently those packages are updated, and sometimes such updates include fixes for critical and timely vulnerabilities.

brada4, 2023-01-21:
If you expand vendor answer to exact post you see that they will never install whatever "official" blob you are demanding here. User is told to be happy with whatever gl.inet packaged, no matter upstream release.

Seems this among other things included improves blocklist sanitation:
https://github.com/AdguardTeam/AdGuardHome/commit/48cbc7bdf0b031725e99dce9f7be23a6df3ff49a

Your approach of removing packaged package and installing upstream static blob will not work in average case - no flash space, no RAM, probably even smaller CPU iSA than supposed.

 Also unlikely there is a way for random file in /overlay to survive sysupgrade and autostart.

CodeSapiens, 2023-01-21:
> If you expand vendor answer to exact post you see that they will never install whatever "official" blob you are demanding here. User is told to be happy with whatever gl.inet packaged, no matter upstream release.
> 
> Seems this among other things included improves blocklist sanitation: [AdguardTeam/AdGuardHome@48cbc7b](https://github.com/AdguardTeam/AdGuardHome/commit/48cbc7bdf0b031725e99dce9f7be23a6df3ff49a)
> 
> Your approach of removing packaged package and installing upstream static blob will not work in average case - no flash space, no RAM, probably even smaller CPU iSA than supposed.
> 
> Also unlikely there is a way for random file in /overlay to survive sysupgrade and autostart.

So here is the exact, word for word exchange you are referring to, from the GL.iNet blog:
**OP question: The Adguard settings page reports an update is available. Is that handled by firmware updates?
GL answer: This is an official update for AdGuardHome, not our firmware update.**

After that, the OP seems quite satisfied with the answer, and the GL rep moves on to other topics, quite nice and friendly.
I am not sure where you read a rebuke implying "just be happy with whatever gl.inet packaged, no matter upstream release". I see no such thing.

I will say this without getting lost in the weeds... I own more than one GL.iNet router, and I have never had an issue with "blocklist sanitation" (whatever that is), or overlapping blobs, or lack of space or RAM, or settings not surviving sysupgrade. In the other hand, my other installations on Raspberry Pi and other Linux test servers, I have had to either install the often obsolete version included in the package, or side load it myself, which then becomes a burden of managing autostart and preserving data to prevent sysupgrade loss. All because the included AGH package is handicapped.

It seems to me that the decision to prevent AGH from checking for updates AFTER INSTALL, both automatically and manually, is being made rather capriciously, and based on speculation rather than actual testing. I have tested both methods quite thoroughly, I believe. Please know that by handicapping the app installation, you are also removing the button that allows users to check for updates on demand. Also know that allowing users the choice to install the latest available update, does not at all require any more RAM or storage, nor does it install in a separate location that will lose its settings upon sysupgrade. It simply upgrades the existing binaries to the latest version. How can that be problematic? And more importantly, have you actually taken the time to test it?

I digress... Peace to you mate.

Spudz76, 2023-01-21:
Flash is slowly damaged by writing stuff to it, so in general most things have any automatic periodic writes disabled.  Such as the logger logs to RAM and does not write anything to flash, to avoid block-burn.

Also a 40MB update is pretty huge when you have limited flash space in /overlay

I know I wouldn't want stuff auto-updating and toasting my flash to death for essentially no reason.

CodeSapiens, 2023-01-21:
> Flash is slowly damaged by writing stuff to it, so in general most things have any automatic periodic writes disabled. Such as the logger logs to RAM and does not write anything to flash, to avoid [...]

Again, removing the --no-check-update on the installer does no such thing, as you would see if you tested. All it does is probe the remote repository for an updated version and compares it to the current installed version, and it only does it AFTER installation is complete, and periodically after that. If a newer version is available IT NOTIFIES THE USER with a friendly hyperlink at the top of the page. It does not go rogue and arbitrarily auto-update anything. Not sure where you got that idea.

Also importantly, it enables a button at the bottom of the page, which the user can click any time to check for updates. Again, none of those two things automatically install anything. Only when the user clicks on the hyperlink does the updated version get installed, and this, I am sure is no different from me going manually and installing it via the CLI or LuCI. Actually, if you want to compare unnecessary writes to flash, think about updating the ENTIRE package manager list, only to see if the developers have gotten around to include the latest version of whatever I'm looking for. Talk about flash toasting.

I think there is a serious misunderstanding here of what exactly the **--no-check-update** parameter does. I don't mean to be confrontational, but I am hearing lots of assumptions and no factual testing evidence.

Anyway, I don't think I'm getting anywhere with this and I have other things to do. Peace.

brada4, 2023-01-21:
If you have storage space you can migrate to the upstream binary with builtin update as in reddit thread. For all other cases opkg works fine.

Spudz76, 2023-01-22:
> probe the remote repository
> and periodically after that
> enables a button

So where does it not-write those results to in order to enable the button?  Probably a single eraseblock which is even worse than larger things which could be wear-levelled (where supported) because they take multiple eraseblocks.

You asked why, not sure why you're mad about the reason.

brada4, 2023-01-24:
Lets  get serious about pinging friendly urls and including friendlier links from them
https://www.google.com/amp/s/techcrunch.com/2018/09/20/adguard-resets-all-user-passwords-after-account-hacks/amp/

diizzyy, 2023-07-30:
Packages needs to be as close to static as possible and most devices don't have space or even a suitable file system / storage backend that allows for frequently changing data.

Closing this as we can consider it as not within the scope of what OpenWrt targets

closed 2023-07-30 by diizzyy