openwrt/packages #20352: PBR: Always miss something when using

PussAzuki, 2023-01-20:
Maintainer: @stangri 
Environment: aarch64 ,Redmi AX6000 stock version, snapshot  (r21861-5dee596501)

Description:
Since yesterday, I officially learned to use PBR and everything felt fine until today when I realized that the system and browser only access IPv4 addresses for some websites. Up until just now, I thought I was searching for the wrong IP range until I connected to the router and took a look at the nftables and realized that some of the IPv6 addresses were not being imported into the sets correctly.

[image]
[image]

You can see that there have been some IPv6 segments that have been ignored.

stangri, 2023-01-20:
Looks like they were merged. 

If you can edit `/etc/init.d/pbr` on your router and remove the `auto-merge;` from [lines 596-608](https://github.com/openwrt/packages/blob/f4f899f6f992db1e92f02881649e1580958dae18/net/pbr/files/etc/init.d/pbr.init#L596-L608) and restart service, does it change things?

PussAzuki, 2023-01-21:
No, they were not merged at all. You can see that there is no possibility of merging the ipv6 segments of google.

[image]
Still the same.

stangri, 2023-01-21:
I have no further insight into this, you can try adding some debug output into nftset function to ensure they it tries to add missing segments, but I have no idea why it silently fails.

You can also try moving the missing segments into a different policy(ies) to see if that works.

PussAzuki, 2023-01-22:
I found that the nftables sets file of dnsmasq is stored in `/var/dnsmasq.d/pbr` , I opened it and saw that some of the addresses are missing. So the question becomes which ones are not translated by default and why.

[image]
As you can see, there are still some IPv6 segments that are being ignored.

PussAzuki, 2023-01-22:
[image]
In this way, I gathered this image:

[image]
[image]

Some pure IPv4/IPv6 segments go through the add process, while others go through the add_dnsmasq_element process.

So it is the location of the problem is found

stangri, 2023-01-22:
Please post complete `/etc/config/pbr` and the output of `service pbr status` after service has ran.

PussAzuki, 2023-01-22:
config：
https://pastebin.com/rw6j4Hz2
status：
https://pastebin.com/RU455cPQ

Public IP address deleted

stangri, 2023-01-22:
@PussAzuki thanks! 

The IPv6 addresses should never end up in the dnsmasq file, I believe the `is_domain()` wasn't properly transferred from `vpn-policy-routing` code and introduced this issue.
If you're willing to test a fix, please install `pbr 1.0.1-16` from [my repo](https://docs.openwrt.melmac.net/) or you can apply [the fix](https://github.com/stangri/source.openwrt.melmac.net/commit/734592dd48ac785aa9a96736d79378205f386da2) directly.

PussAzuki, 2023-01-22:
I tried the latest version and it looks like he's separating out the ipv6 addresses properly. 
Thanks a lot.

PussAzuki, 2023-01-23:
[image]
[image]
ipv4 routing has been handled well...but
[image]
[image]
there is no routing for ipv6 (the rules are there), maybe it needs to be fixed all together?

I have added masq6 to the vpn zone according to the openwrt.org documentation, and IPv6 NAT works fine when using the static routing function under Network - Routing, so there is still a problem to be solved...

stangri, 2023-01-23:
You may want to open a new issue dedicated to IPv6 routing and provide the configs (not screenshots) requested in the Help section of the README, as this will be automatically closed when I merge the linked PR.

closed 2023-01-23 by stangri