openwrt/packages #20354: SmartDNS: Doesn't take into account dhcp 'Hostnames'

ice-fly, 2023-01-20:
Not certain if this does here, in the smart dns repo, or in the luci repo. It's kinda an issue of the package not shaking hands with luci...

**Maintainer:** @pymumu 
**Environment:** _OpenWrt 22.03.3, r20028-43d71ad93e_
**Description:**
I have a 2 openwrt routers on the same network... one in a [dumb ap](https://openwrt.org/docs/guide-user/network/wifi/dumbap) configuration the other running smartdns. On the router running smart DNS I have a few hostnames conveniently added so I can ssh into a short hostname rather than a long ipv4 address:
[image]

When I use stock DNSmasq, the hostnames resolve and I can access the devices. When I turn on SmartDNS, it cannot resolve those names, instead returning: ```DNS_PROBE_FINISHED_NXDOMAIN```
I tried forcing it by using the domain addresses configuration: ```address /example.com/1.2.3.4``` Maybe i eff'ed this up?

Thanks for developing this codebase in general!

Spudz76, 2023-01-21:
```
user@host:/usr/src/openwrt$ grep -ir hosts feeds/packages/net/smartdns/
user@host:/usr/src/openwrt$
```

Would appear it doesn't read any `hosts` file, there is no mention in the official SmartDNS docs either.

ice-fly, 2023-01-21:
I should post in [the feature requests for SmartDNS repo](https://github.com/pymumu/smartdns/issues/1236) and probably close this here?

pymumu, 2023-01-22:
smartdns does not have the ability to read the hosts file, but it can resolve the local host name in the following two ways:   
1. Add 'address /hostname/ip' in
Custom configuration in the smartdns
 UI to resolve the hostname. 
2. Use the ability of smartdns to read the dnsmasq lease file to auto
 resolve the host name. For details, see 1.2 of the FAQ in readme_en.md. This function is configurable in the smartdns advanced settings and is enabled by default.

Spudz76, 2023-01-22:
> I should post in [the feature requests for SmartDNS repo](https://github.com/pymumu/smartdns/issues/1236) and probably close this here?

One would assume since it's so obvious that if they wanted it would already be that way.  The smartdns docs suggest you can setup both dnsmasq and smartdns so that one includes the other, which would effectively restore the `/etc/hosts` feature because dnsmasq supports it.  Essentially you run dnsmasq on some other port than 53 while smartdns is the main resolver but then set smartdns to query the custom dnsmasq port (along with whatever other external servers).  It would still return either the only positive response, or the fastest.