openwrt/packages #20369: Wake on Lan

postlgue, 2023-01-22:
https://github.com/openwrt/luci/issues does not work

 My system: OpenWrt SNAPSHOT (r21757-895f38ca1e) , TP-Link Archer A7 v5

luci-app-wol	git-21.154.28269-c7b7b42	2.68 KiB	LuCI Support for Wake-on-LAN

Services/Wake on LAN seems to show only running devices at my home network  in the dropdown list for choosing targets. 
It does not  remember earlier chosen targets.

I have to enter a custom MAC adress every time.

By the way I think it is useless to show already runnig devices at all. 

I tried to change  /etc/config/etherwake as follows, but no succcess.

config 'etherwake' 'setup'
	# possible program paths
	option 'pathes' '/usr/bin/etherwake /usr/bin/ether-wake'
	# use sudo, defaults to off
	option 'sudo' 'off'
	# interface, defaults to 'eth0'
	# -i <ifname>
	option 'interface' 'br-lan'
	# send wake-up packet to the broadcast address, defaults to off
	# -b
	option 'broadcast' 'off'

config 'target'
	# name for the target
	option 'name' 'xxxxx'
	# mac address to wake up
	option 'mac' 'xx:xx:xx:xx:xx'
	# password in hex without any delimiters
	option 'password' 'xxxxxxxxxxxxxxxxx'
	# wake up on system start, defaults to off
	option 'wakeonboot' 'off'

Is there any possibility to change this behavior.

brada4, 2023-02-02:
Please repost this clearly LuCI issue https://github.com/openwrt/luci/issues/new

diizzyy, 2023-07-30:
Closing, submitted to wrong repo

closed 2023-07-30 by diizzyy