openwrt/packages #20374: PBR: No working ipv6 tables

PussAzuki, 2023-01-23:
Maintainer: @stangri 
Environment: aarch64 ,Redmi AX6000 stock version, snapshot (r21861-5dee596501)
(Temporarily not updated, I want to wait until after this PR merger to update)

Description:

No IPv6 routing, even if ipv6 support is turned on.

https://pastebin.com/rw6j4Hz2 (No change in configuration compared to #20373 )

https://pastebin.com/RDUVAvuY (You can see that there is still no ipv6 table relative to the status in the last pr)

I have added masq6 to the vpn zone according to the openwrt.org documentation, and IPv6 NAT works fine when using the static routing function under luci Network - Routing, so the firewall should be fine.

PussAzuki, 2023-01-24:
https://pastebin.com/TVaYpf4N

I really think it would be better to use ifconfig to see IPv6 addresses...

stangri, 2023-01-25:
> I really think it would be better to use ifconfig to see IPv6 addresses...

ifconfig is deprecated.

PussAzuki, 2023-01-25:
On my side, pppoe-wan has one 240e:aaaa:xxxx/128 IPv6 adress and one 240e:bbbb:yyyy/60 IPv6 PD prefix, and wireguard has one 172.16.0.2 IPv4 adress and one 2606:cccc:zzzz/128 IPv6 adress.

I assign the /64 of ipv6 pd to lan and let it assign ipv6 address to the device automatically.

Now I make the ipv6 routing for each device under the router mainly through ppoe-wan instead of NAT, and then tunnel to IPv6 NAT with wireguard.

I've also confirmed the possibility of going this way, does PBR support this?