openwrt/packages #20391: banIP: tor isn't maintained anymore and list is empty

realizelol, 2023-01-25:
Maintainer: @dibdot + @rg4github
Environment: openWRT 21.02.5

Description:
------------
The tor lists are empty:
- https://lists.fissionrelays.net/tor/exits-ipv4.txt
- https://lists.fissionrelays.net/tor/exits-ipv6.txt
and have been deprecated as the maintainer stopped working on the project since 2022-11-04.
Read more on here: https://fissionrelays.net/ 
You could also check that the lists inside there are all 0 byte files: https://lists.fissionrelays.net/tor/
And the ipsets tor_4 and tor_6 will not being created and on any change it will being updated.
If the fetch wasn't successful it will not been overridden and the old list will remain.

Please update lists @ Line 158 + 159:
https://github.com/openwrt/packages/blob/master/net/banip/files/banip.sources#L158-L159
to example my lists (which will check the torproject.org API every day @ 00:00 UTC:
- https://raw.githubusercontent.com/realizelol/torblocklist/master/exits-ipv4.txt
- https://raw.githubusercontent.com/realizelol/torblocklist/master/exits-ipv6.txt

<br>

best regards
realizelol

brada4, 2023-01-25:
There is one more official at https[:]//check.torproject.org/torbulkexitlist

realizelol, 2023-01-25:
This is a good objection but for me it was down for hours on yesterday which will make this unattractive.
While the REST API (onionoo) was still online. And also there are actually just 1245 lines instead of 1345:
```
curl -s https://check.torproject.org/torbulkexitlist | wc -l
    1235
curl -s https://raw.githubusercontent.com/realizelol/torblocklist/master/exits-ipv4.txt | wc -l
    1345
```
This is caused there are only exit_addresses and nor or_addresses which should also - especially on IPv6 - be blocked.
Some official explanations:
>or_addresses [onionoo](https://metrics.torproject.org/onionoo.html#details_relay_or_addresses)
Array of IPv4 or IPv6 addresses and TCP ports or port lists where the relay accepts onion-routing connections. The first address is the primary onion-routing address that the relay used to register in the network, subsequent addresses are in arbitrary order. IPv6 hex characters are all lower-case.

>exit_addresses [onionoo](https://metrics.torproject.org/onionoo.html#details_relay_exit_addresses)
Array of IPv4 addresses that the relay used to exit to the Internet in the past 24 hours. Omitted if array is empty.

<br/>

@dibdot is also allowed to fork it or create a list generator by it's own as he has already done for DoH:
https://github.com/openwrt/packages/blob/master/net/banip/files/banip.sources#L41-L42

brada4, 2023-01-26:
First lists are from exits, i.e IP-s that connects to your webserver. Blocking entry points borders censorship and you did not ask for it.

dibdot, 2023-01-26:
banIP 0.7.x in branch 21.02. is no longer maintained by me ... 
Feel free to open a proper and well run tested pull request to change the download location for tor nodes in banIP 0.7.x/21.02. Currently I'm working on a nftables port (0.8.x, not yet available in the official repo) which includes the following tor feed: https://raw.githubusercontent.com/SecOps-Institute/Tor-IP-Addresses/master/tor-exit-nodes.lst

realizelol, 2023-01-27:
PRs for "openwrt-21.02" + "22.03" have been created. So I'm closing this issue and waiting for further instructions or merge in PRs. Glad to hear you're already working on an nft version.

closed 2023-01-27 by realizelol