openwrt/packages #20413: libxml2: cmake error: Could NOT find Iconv

douo, 2023-01-28:
Maintainer: @neheb 
Environment: Arch Linux x86_64 22.03.3

Description:

```
CMake Error at xxx/staging_dir/host/share/cmake-3.22/Modules/FindPackageHandleStandardArg
s.cmake:230 (message):
  Could NOT find Iconv (missing: Iconv_LIBRARY)
Call Stack (most recent call first):
xxx/staging_dir/host/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:594 (
_FPHSA_FAILURE_MESSAGE)
  xxx/staging_dir/host/share/cmake-3.22/Modules/FindIconv.cmake:165 (find_package_handle_
standard_args)
  CMakeLists.txt:67 (find_package)
```

Try: 

- <del>compile libconv-full</del>
- ~~install libiconv on host~~

neheb, 2023-01-28:
Is full NLS enabled?

douo, 2023-01-28:
I have try:
```
CONFIG_BUILD_NLS=y
```
Issue still exist.

neheb, 2023-01-28:
Oh this is 22.03. why was this backported?

You need https://github.com/openwrt/openwrt/commit/2cb48f58c247c801a101b1625df2ff71a4f8be71

douo, 2023-01-29:
Oh Thanks. It's my fault. I misconfigured the feeds.

closed 2023-01-29 by douo