openwrt/packages #20419: usteer: parameters for node BSS transition seems not to be right

McGiverGim, 2023-01-30:
Maintainer: @blocktrron 
Environment: ipq807, Xiaomi AX3600, master

Description:

I've been trying to use usteer for making my phones roam between nodes, without luck. Using "manual" ubus commands like this:
```
ubus call hostapd.wlan5g bss_transition_request '{"addr":"XXX", "disassociation_imminent":true, "disassociation_timer":100}'
```
the change between nodes is produced. Looking at `usteer` code it seems `disassociation_imminent` is `false` and it does not use `disassociation_timer`. Without any of this parameters my phones do not change from node.
I don't know if this parameters are in this way for something, but at least a config option to change them will fix the issue for me and others, as I've seen in the forums.

McGiverGim, 2023-01-31:
I've been doing more tests: it seems it works with the `disassociation_imminent` as `false` if I pass all the information like this:
```
ubus call hostapd.wlan5g bss_transition_request '{"addr": "f0:5c:77:xx:xx:xx", "disassociation_imminent": false, "di
sassociation_timer": 10, "validity_period": 30, "neighbors": ["xxxxxxxxxxxxxx"], "abridged": true }'
```

xxsamxx1991, 2023-01-31:
I can also attest to this. The same problem is evident for my android devices.

McGiverGim, 2023-02-01:
Doing more tests, with `disassociation_imminent` seems that is up to the client to do the roaming. Some of my clients do not roam, If I put `disassociation_imminent` as `true` then the AP seems to disconnect it and the client then connects again to the other AP. 
Maybe some parameter to "force" after X time if the client does not roam will be great.

McGiverGim, 2023-02-08:
I close this. After looking deeply at the code, it seems the transition request is done with the correct parameters, so I will do more tests to see if there is some bug or a bad configuration on my own.

closed 2023-02-08 by McGiverGim

Mushoz, 2024-05-23:
What turned out to be the missconfiguration? I am only seeing the following messages in my log:

`daemon.notice hostapd: wl0-ap0: BSS-TM-RESP xx:xx:xx:xx:xx:xx status_code=1 bss_termination_delay=0` 

And band steering does not seem to work

Mushoz, 2024-05-23:
It looks like the neighbors list must be in the bss transition command. How do I get usteer to do that?