openwrt/packages #20424: dnsmasq

iopen2, 2023-01-31:
# Issue template (remove lines from top till here)

Specific package problem detected in :
https://downloads.openwrt.org/releases/packages-21.02/aarch64_generic/base/dnsmasq-dhcpv6_2.85-9_aarch64_generic.ipk

Network Topology :
Gateway : openwrt 19.07.6, dnsmasq-dhcpv6 2.80-16.3, LAN ipaddr 172.25.5.1/24
LAN client : openwrt 21.02, dndmasq-dhcpv6 2.85-9, LAN ipaddr 172.25.5.71/24

Description:
Client has a static IP address and is using the default dnsmasq configuration.
Client nslookup returns no data when the queried name is in the gateway's /etc/hosts
Wireshark capture shows that the gateway's dnsmasq does send a response containing the IP address.
Client nslookup returns the expected address(es) in other cases 
No other hosts on the LAN experience the problem.

The following examples were captured by wireshark remotely from tcpdump running on the client's LAN interface.  They refer to the capture file which github has refused to accept - either as .zip or .tgz - we can/will supply if a suitable method can be found.

1: Client action : Produces wireshark rows 1..4
   ====================================
root@ls01:~# nslookup dnsmasq.bug
Server:         127.0.0.1
Address:        127.0.0.1:53

Non-authoritative answer:

2: Client action : Produces wireshark rows 5..8
   ====================================
root@ls01:~# nslookup dnsmasq.bug 172.25.5.1
Server:         172.25.5.1
Address:        172.25.5.1:53

Name:   dnsmasq.bug
Address: 172.20.42.42

Non-authoritative answer:

3: Client action : Produces wireshark rows 9..12
   =====================================
root@ls01:~# nslookup openwrt.org
Server:         127.0.0.1
Address:        127.0.0.1:53

Non-authoritative answer:
Name:   openwrt.org
Address: 139.59.209.225

Non-authoritative answer:
Name:   openwrt.org
Address: 2a03:b0c0:3:d0::1af1:1

brada4, 2023-01-31:
Wireshark capture seems absent?
Is your ISP in violation of https://www.icann.org/en/system/files/files/sac-032-en.pdf for commercial gain?

BKPepe, 2023-03-16:
Seems like I am closing this due to lack of OP's response.