openwrt/packages #20440: attendedsysupgrade-common

atomGit, 2023-02-04:
# Issue template (remove lines from top till here)

Maintainer:  @aparcar 
Environment: ARM, Linksys WRT1900ACS, OpenWrt 21.02.5

Description:

i believe the newest stable firmware version as of this date is 22.03.2, however attendedsysupgrade does not seem to find an upgrade unless the advanced option (Luci) is enabled

olimpiumarius, 2023-02-05:
wrt1900acs v2 both images Factory and Sysupgrade fail to install 
[](url)

olimpiumarius, 2023-02-05:
`Building images for mvebu - Linksys WRT1900ACS v1
Packages: 6in4 adblock ath10k-board-qca988x ath10k-firmware-qca988x-ct attendedsysupgrade-common auc base-files block-mount busybox ca-bundle ca-certificates cgi-io collectd collectd-mod-conntrack collectd-mod-contextswitch collectd-mod-cpu collectd-mod-cpufreq collectd-mod-df collectd-mod-disk collectd-mod-dns collectd-mod-filecount collectd-mod-interface collectd-mod-iptables collectd-mod-irq collectd-mod-iwinfo collectd-mod-load collectd-mod-memory collectd-mod-netlink collectd-mod-network collectd-mod-openvpn collectd-mod-ping collectd-mod-processes collectd-mod-protocols collectd-mod-rrdtool collectd-mod-tcpconns collectd-mod-thermal collectd-mod-uptime collectd-mod-vmem collectd-mod-wireless confuse curl ddns-scripts ddns-scripts-services dnsmasq-full drill dropbear e2fsprogs firewall4 fstools fwtool getrandom hostapd-common htop ip-tiny ip6tables-nft iperf3 ipset iptables-mod-conntrack-extra iptables-mod-ipopt irqbalance iw iwinfo jshn jsonfilter kmod-ata-core kmod-ath kmod-ath10k-ct kmod-ath9k kmod-ath9k-common kmod-bluetooth kmod-btmrvl kmod-cfg80211 kmod-crypto-acompress kmod-crypto-aead kmod-crypto-authenc kmod-crypto-ccm kmod-crypto-cmac kmod-crypto-crc32c kmod-crypto-ctr kmod-crypto-ecb kmod-crypto-ecdh kmod-crypto-gcm kmod-crypto-gf128 kmod-crypto-ghash kmod-crypto-hash kmod-crypto-hmac kmod-crypto-kpp kmod-crypto-lib-chacha20 kmod-crypto-lib-chacha20poly1305 kmod-crypto-lib-curve25519 kmod-crypto-lib-poly1305 kmod-crypto-manager kmod-crypto-null kmod-crypto-rng kmod-crypto-seqiv kmod-crypto-sha256 kmod-cryptodev kmod-fs-exportfs kmod-fs-ext4 kmod-fs-vfat kmod-fs-xfs kmod-gpio-button-hotplug kmod-hid kmod-hwmon-core kmod-hwmon-drivetemp kmod-hwmon-gpiofan kmod-hwmon-nct7802 kmod-i2c-core kmod-i2c-mux kmod-i2c-mux-pca954x kmod-ifb kmod-input-core kmod-input-evdev kmod-ip6tables kmod-ipt-conntrack kmod-ipt-conntrack-extra kmod-ipt-core kmod-ipt-ipopt kmod-ipt-ipset kmod-ipt-nat kmod-ipt-offload kmod-ipt-raw kmod-leds-turris-omnia kmod-lib-crc-ccitt kmod-lib-crc16 kmod-lib-crc32c kmod-lib-lzo kmod-linkstation-poweroff kmod-mac80211 kmod-md-mod kmod-md-raid0 kmod-md-raid1 kmod-mii kmod-mmc kmod-mwifiex-sdio kmod-mwlwifi kmod-nf-conntrack kmod-nf-conntrack-netlink kmod-nf-conntrack6 kmod-nf-flow kmod-nf-ipt kmod-nf-ipt6 kmod-nf-log kmod-nf-log6 kmod-nf-nat kmod-nf-reject kmod-nf-reject6 kmod-nfnetlink kmod-nft-compat kmod-nft-core kmod-nft-fib kmod-nft-nat kmod-nft-offload kmod-nls-base kmod-nls-cp437 kmod-nls-iso8859-1 kmod-nls-utf8 kmod-ppp kmod-pppoe kmod-pppox kmod-regmap-core kmod-regmap-i2c kmod-rtc-rs5c372a kmod-rtc-s35390a kmod-sched-cake kmod-sched-core kmod-scsi-core kmod-slhc kmod-tun kmod-udptunnel4 kmod-udptunnel6 kmod-usb-core kmod-usb-ledtrig-usbport kmod-usb-net kmod-usb-storage kmod-usb-xhci-hcd kmod-usb-xhci-pci-renesas kmod-usb2 kmod-usb3 kmod-wireguard libblkid libblobmsg-json libcap-ng libcomerr libcurl libelf libext2fs libf2fs libip4tc libip6tc libiperf3 libipset libiptext libiptext-nft libiptext6 libiwinfo libiwinfo-data libiwinfo-lua libjson-c libjson-script libldns liblua liblucihttp liblucihttp-lua liblucihttp-ucode libmnl libncurses libnftnl libnl-core libnl-route libnl-tiny libopenssl libopenssl-conf libopenssl-devcrypto libpcap libpcre libpthread libqrencode librt libsmartcols libss libubox libubus libubus-lua libuci libuci-lua libuclient libucode libustream-openssl libuuid libuv libwolfssl libxtables logd lua luci luci-app-adblock luci-app-advanced-reboot luci-app-attendedsysupgrade luci-app-ddns luci-app-firewall luci-app-nlbwmon luci-app-openvpn luci-app-opkg luci-app-sqm luci-app-statistics luci-app-upnp luci-app-wifischedule luci-base luci-compat luci-lib-base luci-lib-ip luci-lib-ipkg luci-lib-json luci-lib-jsonc luci-lib-nixio luci-mod-admin-full luci-mod-dashboard luci-mod-network luci-mod-rpc luci-mod-status luci-mod-system luci-proto-ipv6 luci-proto-ppp luci-proto-wireguard luci-ssl-openssl luci-theme-bootstrap luci-theme-material luci-theme-openwrt-2020 miniupnpd-nftables mkf2fs mtd mtr-json mwifiex-sdio-firmware mwlwifi-firmware-88w8864 mwlwifi-firmware-88w8964 nano netifd nftables nftables-json odhcp6c odhcpd-ipv6only openssl-util openvpn-mbedtls openwrt-keyring opkg partx-utils ppp ppp-mod-pppoe ppp-mod-pptp procd procd-seccomp procd-ujail px5g-wolfssl qrencode rpcd rpcd-mod-file rpcd-mod-iwinfo rpcd-mod-luci rpcd-mod-rpcsys rpcd-mod-rrdns rpcd-mod-ucode screen sqm-scripts tc-mod-iptables tc-tiny tcpdump terminfo ubi-utils uboot-envtools ubox ubus ubusd ucert uci uclient-fetch ucode ucode-mod-fs ucode-mod-html ucode-mod-math ucode-mod-ubus ucode-mod-uci uhttpd uhttpd-mod-ubus urandom-seed urngd usign wget-ssl wifischedule wireguard-tools wireless-regdb wpad-basic-mbedtls xtables-nft zlib base-files busybox ca-bundle dropbear firewall4 fstools iwinfo kernel kmod-gpio-button-hotplug kmod-mwlwifi kmod-nft-offload logd mtd mwlwifi-firmware-88w8864 netifd nftables odhcp6c odhcpd-ipv6only opkg ppp ppp-mod-pppoe procd procd-seccomp procd-ujail uboot-envtools uci uclient-fetch urandom-seed urngd wpad-basic-mbedtls

Installing packages...
Installing libc (1.2.3-4) to root...
Installing libgcc1 (12.2.0-4) to root...
Copying /home/aparcar/asu/worker1/cache/SNAPSHOT/mvebu/cortexa9/dl/libgcc1_12.2.0-4_arm_cortex-a9_vfpv3-d16.ipk.
Configuring libgcc1.
Configuring libc.
Installing kernel (5.15.91-1-a659a6adaf8a9973d2bc741a3eb765c1) to root...
Configuring kernel.
Installing 6in4 (28) to root...
[... cut ...]
sysupgrade-armada-385-linksys-shelby/
sysupgrade-armada-385-linksys-shelby/CONTROL
sysupgrade-armada-385-linksys-shelby/kernel
sysupgrade-armada-385-linksys-shelby/root
[rootfs]
mode=ubi
vol_id=0
vol_type=dynamic
vol_name=rootfs
image=/home/aparcar/asu/worker1/cache/SNAPSHOT/mvebu/cortexa9/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/root.squashfs
vol_size=11657216
[rootfs_data]
mode=ubi
vol_id=1
vol_type=dynamic
vol_name=rootfs_data
vol_size=1MiB
vol_flags=autoresize

Calculating checksums...`

atomGit, 2023-02-05:
why did you post in this issue? your issue is very different - you need to open a *new* issue