openwrt/packages #20441: borgbackup: missing modules on Linksys WRT1200AC (armv7l) with OpenWrt 22.03.0

snpz, 2023-02-04:
Maintainer: @julienmalik 
Environment: armv7l, Linksys WRT1200AC, openwrt 22.03.0

Description:
I've installed the package borgbackup via Luci on my router, but when I run the borg command afterwards I'm receiving the following errors:

```
root@router:~# borg
Traceback (most recent call last):
  File "/usr/bin/borg", line 10, in <module>
    from importlib.metadata import distribution
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 6, in <module>
ModuleNotFoundError: No module named 'email'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/borg", line 13, in <module>
    from importlib_metadata import distribution
ModuleNotFoundError: No module named 'importlib_metadata'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/borg", line 15, in <module>
    from pkg_resources import load_entry_point
ModuleNotFoundError: No module named 'pkg_resources'
```

Any suggestions what I need to do to make it work?

Thanks in advance!

julienmalik, 2023-02-07:
The package is probably simply missing some dependencies. And I probably did not notice since the system on which I test already have them installed.

From your logs you can first try :
- `python3-email`
- `python3-pkg-resources`

snpz, 2023-02-07:
I just tried your suggestions, but there wasn't enough free space on my router for the installation. I reflashed without my usual additional packages, taking just the suggested ones in firmware selector to test it.

Both packages you mentioned were needed and additionally `python3-packaging`. When running the `borg` command I'm now getting the expected help output. (did not test any further)

Thanks for your help, unfortunately I will have to switch to something else because of my free space limitations. (probably restic)

julienmalik, 2023-02-07:
I tested on a x86 docker and after installing the package, had to install :
- `python3-email`
- `python3-urllib`
- `python3-packaging`
- `python3-uuid`
- `python3-unittest`

some of them clearly miss from the package dependency list and I should fix this, but the first two look like missing dependencies when using the importlib package.

PR #20463 (2023-03-27): borgbackup: fix missing depends
Maintainer: me
Compile tested: no
Run tested: yes (x86 docker)

Description:

Fixes #20441
Workaround for #20462 

```
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r22000-6bc675c0be
 -----------------------------------------------------
=== WARNING! =====================================
 [... cut]
commit 1f25be97 (2023-03-27): borgbackup: add missing dependencies

The initial package submission was missing
some required and optional dependencies
due to lack of testing on a system without any python
related packages pre-installed.

Some optional but highly recommended dependencies
were discovered with the stdlib module as described in:
https://github.com/openwrt/packages/blob/392a68e24774294590abf9c08ea1832f2cee190d/lang/python/README.md

Fixes #20441
commit 31f36e5b (2023-04-12): borgbackup: add missing dependencies

The initial package submission was missing
some required and optional dependencies
due to lack of testing on a system without any python
related packages pre-installed.

Some optional but highly recommended dependencies
were discovered with the stdlib module as described in:
https://github.com/openwrt/packages/blob/392a68e24774294590abf9c08ea1832f2cee190d/lang/python/README.md

Fixes #20441

(cherry picked from commit 1f25be97b6c73691ec5a00a9deb9a7456e59be8c)
closed 2023-03-27 by BKPepe