openwrt/packages #20452: samba4: cannot connect due to permission denied on share_info.tdb

gfrancesco, 2023-02-06:
Maintainer: @Gingernut1978 
Environment: ARMv5TE, ZyXEL NSA325-Feroceon 88FR131, OpenWrt SNAPSHOT r21997-b5193291bd

Description:
I just installed a fresh snapshot of OpenWrt and I run into a new problem while trying to connect to my smb shares.
The connection is refused and I found this in `/var/log/log.rpcd_classic`

```
[2023/02/06 00:26:56,  0] ../../source3/rpc_server/rpc_worker.c:1105(rpc_worker_main)
  rpcd_classic version 4.17.2 started.
  Copyright Andrew Tridgell and the Samba Team 1992-2022
[2023/02/06 00:26:57,  0] ../../source3/lib/sharesec.c:161(share_info_db_init)
  Failed to open share info database /var/lib/samba/share_info.tdb (Permission denied)
[2023/02/06 00:26:57,  0] ../../source3/lib/sharesec.c:161(share_info_db_init)
  Failed to open share info database /var/lib/samba/share_info.tdb (Permission denied)
[2023/02/06 00:26:57,  0] ../../source3/lib/sharesec.c:161(share_info_db_init)
  Failed to open share info database /var/lib/samba/share_info.tdb (Permission denied)
[2023/02/06 00:26:57,  0] ../../source3/lib/sharesec.c:161(share_info_db_init)
  Failed to open share info database /var/lib/samba/share_info.tdb (Permission denied)
[2023/02/06 00:26:57,  0] ../../source3/lib/sharesec.c:161(share_info_db_init)
  Failed to open share info database /var/lib/samba/share_info.tdb (Permission denied)
``` 

The permissions on `/var/lib/samba/share_info.tdb` are set to `rw` for root only, if I add `rw` for other users and restart the service I can connect and it works as usual.
If I restart the system the permissions are back to  `rw` for root only, I guess the files in `/var/lib/samba/` are re-created at boot.
I'm on `samba4-server` v4.17.2-2

This was not happening on `samba4-server` v4.14.12-1, OpenWrt SNAPSHOT r18717-0e32c6baf3.
I preserved the old configuration. I backed up and copied to the new installation the following files:

```
/etc/config/samba4
/etc/samba/secrets.tdb
/etc/samba/smb.conf.template
/etc/samba/smbpasswd
/etc/avahi/services/smb.service
```

Neustradamus, 2023-05-18:
@gfrancesco: Have you tried to use ksmbd?

Neustradamus, 2023-05-18:
@jmarcet, @neheb, @dhewg, @Gingernut1978, @1715173329, @mpeleshenko, @jjm2473, @1715173329, @PolynomialDivision, @Andy2244, @wigyori, @svlobanov, @champtar, @selanf, @hnyman, @lipnitsk, @ja-pa, @jefferyto, @dbermond, @NoTengoBattery, @diizzyy, @val-kulkov, @dengqf6, @cotequeiroz: Have you seen this ticket?

selanf, 2023-05-21:
@Neustradamus I use less as possivle services/daemons with openwrt anymore. Most of them have problems because of bad coded scripts ... as you noticed

jjm2473, 2023-05-26:
@Neustradamus No, until you mentioned