openwrt/packages #20465: miniupnp doesn't work

PeakGymnast, 2023-02-08:
luci-app-upnp package version：git-22.250.64651-a61c041
Openwrt version：OpenWrt 22.03.3 r20028-43d71ad93e
Device：phicomm k2p

Miniupnp log

`Wed Feb  8 03:54:48 2023 daemon.info procd: Instance miniupnpd::instance1 pid 5763 not stopped on SIGTERM, sending SIGKILL instead
Wed Feb  8 03:54:48 2023 daemon.info miniupnpd[5872]: system uptime is 1428 seconds
Wed Feb  8 03:54:48 2023 daemon.info miniupnpd[5872]: mnl_socket bound, port_id=5872
Wed Feb  8 03:54:48 2023 daemon.info miniupnpd[5872]: Reloading rules from lease file
Wed Feb  8 03:54:48 2023 daemon.info miniupnpd[5872]: version 2.3.0 starting NAT-PMP/PCP UPnP-IGD ext if pppoe-wan BOOTID=1675828488
Wed Feb  8 03:54:48 2023 daemon.info miniupnpd[5872]: specific IPv6 ext if pppoe-wan
Wed Feb  8 03:54:48 2023 daemon.info miniupnpd[5872]: STUN: Performing with host=stun.miwifi.com and port=3478 ...
Wed Feb  8 03:54:48 2023 daemon.debug miniupnpd[5872]: resolve_stun_host: stun.miwifi.com:3478 => 111.206.174.2:3478
Wed Feb  8 03:54:48 2023 daemon.info miniupnpd[5872]: perform_stun: local ports 33530 46357 49001 43161
Wed Feb  8 03:54:48 2023 daemon.debug miniupnpd[5872]: wait_for_stun_responses: waiting 3 secs and 0 usecs
Wed Feb  8 03:54:48 2023 daemon.debug miniupnpd[5872]: wait_for_stun_responses: received responses: 1
Wed Feb  8 03:54:48 2023 daemon.debug miniupnpd[5872]: wait_for_stun_responses: waiting 3 secs and 0 usecs
Wed Feb  8 03:54:51 2023 daemon.debug miniupnpd[5872]: wait_for_stun_responses: select(): no more responses
Wed Feb  8 03:54:51 2023 daemon.debug miniupnpd[5872]: wait_for_stun_responses: waiting 3 secs and 0 usecs
Wed Feb  8 03:54:54 2023 daemon.debug miniupnpd[5872]: wait_for_stun_responses: select(): no more responses
Wed Feb  8 03:54:54 2023 daemon.debug miniupnpd[5872]: wait_for_stun_responses: waiting 3 secs and 0 usecs
Wed Feb  8 03:54:57 2023 daemon.debug miniupnpd[5872]: wait_for_stun_responses: select(): no more responses
Wed Feb  8 03:54:57 2023 daemon.debug miniupnpd[5872]: table_cb(0x7f821340, 0x7f822370) fw4 upnp_forward 2
Wed Feb  8 03:54:57 2023 daemon.debug miniupnpd[5872]: table_cb(0x7f821534, 0x7f822370) fw4 upnp_forward 2
Wed Feb  8 03:54:57 2023 daemon.debug miniupnpd[5872]: table_cb(0x7f821734, 0x7f822370) fw4 upnp_forward 2
Wed Feb  8 03:54:57 2023 daemon.debug miniupnpd[5872]: table_cb(0x7f821934, 0x7f822370) fw4 upnp_forward 2
Wed Feb  8 03:54:57 2023 daemon.debug miniupnpd[5872]: table_cb(0x7f821340, 0x7f822370) fw4 upnp_forward 2
Wed Feb  8 03:54:57 2023 daemon.debug miniupnpd[5872]: table_cb(0x7f821534, 0x7f822370) fw4 upnp_forward 2
Wed Feb  8 03:54:57 2023 daemon.debug miniupnpd[5872]: table_cb(0x7f821734, 0x7f822370) fw4 upnp_forward 2
Wed Feb  8 03:54:57 2023 daemon.debug miniupnpd[5872]: table_cb(0x7f821340, 0x7f822370) fw4 upnp_forward 2
Wed Feb  8 03:54:57 2023 daemon.debug miniupnpd[5872]: table_cb(0x7f821534, 0x7f822370) fw4 upnp_forward 2
Wed Feb  8 03:54:57 2023 daemon.debug miniupnpd[5872]: table_cb(0x7f821340, 0x7f822370) fw4 upnp_forward 2
Wed Feb  8 03:54:57 2023 daemon.debug miniupnpd[5872]: parse_stun_response: Type 0x0101, Length 48, Magic Cookie 2112a442
Wed Feb  8 03:54:57 2023 daemon.debug miniupnpd[5872]: parse_stun_response: MAPPED-ADDRESS 117.152.166.33:38193
Wed Feb  8 03:54:57 2023 daemon.debug miniupnpd[5872]: parse_stun_response: RESPONSE-ORIGIN 111.206.174.2:3478
Wed Feb  8 03:54:57 2023 daemon.debug miniupnpd[5872]: parse_stun_response: OTHER-ADDRESS 111.206.174.3:3479
Wed Feb  8 03:54:57 2023 daemon.debug miniupnpd[5872]: parse_stun_response: XOR-MAPPED-ADDRESS 117.152.166.33:38193
Wed Feb  8 03:54:57 2023 daemon.notice miniupnpd[5872]: perform_stun: 1 response out of 4 received
Wed Feb  8 03:54:57 2023 daemon.notice miniupnpd[5872]: perform_stun: #0 external address or port changed
Wed Feb  8 03:54:57 2023 daemon.warn miniupnpd[5872]: STUN: ext interface pppoe-wan with private IP address 100.64.169.90 is now behind restrictive or symmetric NAT with public IP address 117.152.166.33 which does not support port forwarding
Wed Feb  8 03:54:57 2023 daemon.warn miniupnpd[5872]: NAT on upstream router blocks incoming connections set by miniupnpd
Wed Feb  8 03:54:57 2023 daemon.warn miniupnpd[5872]: Turn off NAT on upstream router or change it to full-cone NAT 1:1 type
Wed Feb  8 03:54:57 2023 daemon.warn miniupnpd[5872]: Port forwarding is now disabled
Wed Feb  8 03:54:57 2023 daemon.notice miniupnpd[5872]: HTTP listening on port 5000
Wed Feb  8 03:54:57 2023 daemon.notice miniupnpd[5872]: HTTP IPv6 address given to control points : [fd7c:160e:e51f::1]
Wed Feb  8 03:54:57 2023 daemon.notice miniupnpd[5872]: Listening for NAT-PMP/PCP traffic on port 5351
Wed Feb  8 03:55:31 2023 daemon.debug miniupnpd[5872]: level=0 type=8
Wed Feb  8 03:55:31 2023 daemon.debug miniupnpd[5872]: ifindex = 10  192.168.1.1`

stintel, 2023-03-10:
You are behind CGNAT. This is not something we can fix. Request your ISP to assign you a public IP instead.

closed 2023-03-10 by stintel

EkkoG, 2023-09-18:
> You are behind CGNAT. This is not something we can fix. Request your ISP to assign you a public IP instead. Also, please don't post logs as screenshots. As for the Xbox showing Open NAT while there is no working port-forward ... Microsoft at its best. I can confirm I am seeing the same on my Xbox Series X. I have tested miniupnpd with upnpc and this works fine, rules are added, and forwarding works. Doing the NAT test in the Xbox settings doesn't create any rules, yet the thing claims open NAT. I have contacted Microsoft about UPnP issues at some point, but my query was completely ignored. So the only advice I can give: stop sending money to Microsoft.

@stintel with [natmap](https://github.com/heiher/natmap) UPnP will work fine when CGNAT type is full-cone

stintel, 2023-09-21:
> @stintel with [natmap](https://github.com/heiher/natmap) UPnP will work fine when CGNAT type is full-cone

Feel free to contribute it. I'm not behind CGNAT so I will not spend time on it.