openwrt/packages #20468: bind: starts with `-4` flag after a router reboot

rucketeg, 2023-02-08:
Maintainer: @nmeyerhans 

Description:
Even with #20434 fixed, bind starts with the `-4` flag after a router reboot. This is probably a race condition - IPv6 (and possibly even IPv4) is not available at the time when `bind` starts. Restarting the `bind` service after `wan6` had enough time to initialize fixes the problem.

What has started in #20183 as a attempt to clean up noise in the logfiles, turned out to be causing denial-of-service for dual-stack and especially IPv6-only environments. Because of that I suggest to rollback c0c312d83b71c7d3d423104aad772a4442090834 and find a better solution to the issue described in #20183. Adding special UCI option for IPv4-only setup seems the most reasonable to me.

The issue could possibly be mitigated by using [procd_add_reload_interface_trigger](https://openwrt.org/docs/guide-developer/procd-init-scripts#specifying_triggers) option of the initscript, but I think this is not correct path - I think that the default configuration should be as simple as possible and applicable to the most common cases.

@pprindeville, @nmeyerhans, @stintel, thoughts?

pprindeville, 2023-03-12:
What's wrong with using `procd_add_reload_interface_trigger`?

rucketeg, 2023-03-16:
Seems like `/etc/init.d/named reload` doesn't pickup the changes to the arguments.

pprindeville, 2023-03-26:
> Seems like `/etc/init.d/named reload` doesn't pickup the changes to the arguments.

Can you dig into that a bit more.  Are you saying that's related to the above issue or in addition to?

pprindeville, 2023-03-29:
> Seems like `/etc/init.d/named reload` doesn't pickup the changes to the arguments.

It's supposed to re-read `/etc/bind/named.conf` (and any files that this includes directly).  What are you referring to by "arguments"?

pprindeville, 2023-07-26:
Looping in @nbd168 @jow- as well...

PR #27471 (2025-09-19): bind: don't break IPv6 support
What started in #20183 as a attempt to clean up noise in the logfiles, turned out to be causing denial-of-service for dual-stack and especially IPv6-only environments.

Breaking core network functionality cannot possibly be less important than cosmetic issues, and those affected by log spam can avoid it via other means (e.g. "query-source-v6 none;" in named.conf).

There's no reliable heuristic for determining whether there's IPv6 connectivity at the time bind is started which will catch any and all corner cases, as discussed in #26327.

So, remove this logic for [... cut]