openwrt/packages #20473: Support python packages that use hatch

lu-zero, 2023-02-09:
`httpx`'s  pyproject.toml apparently is not something the current approach cannot consume:

```
ValueError: invalid pyproject.toml config: `project.license`.
configuration error: `project.license` must be valid exactly by one definition (2 matches found):

    - keys:
        'file': {type: string}
      required: ['file']
    - keys:
        'text': {type: string}
      required: ['text']
```

ofek, 2023-02-13:
FYI this comes from [PEP 639](https://peps.python.org/pep-0639/) which is on the path to being officially accepted but the author lacks free time currently.

lu-zero, 2023-02-13:
I ended up calling `pip install` and it sort of worked as intended.

jefferyto, 2023-05-19:
Real support for pyproject.toml-based builds was added in #20801. This includes packages that use hatchling, like [jsonschema](https://github.com/openwrt/packages/blob/142bbc77f1aa8a81cd1ffb1ebad3ce4f2ef24b8a/lang/python/python-jsonschema/Makefile#L14). For [httpx](https://github.com/encode/httpx/blob/0.24.1/pyproject.toml#L2) only `python-hatchling/host` and `python-hatch-fancy-pypi-readme/host` would be necessary.

closed 2023-05-19 by jefferyto