openwrt/packages #20479: adblock - feature request - fast whitelist refresh

mavack, 2023-02-12:
# Issue template (remove lines from top till here)

Maintainer: @dibdot 
Environment:  Openwrt 22.03.3 / adblock 4.1.5 /rpi4

Description:
Request the ability to fast apply white list via using existing block file > 
grep -v -f adb.whitelist  adb_overall
reload dns as normal.

Or fast regenerate from downloads if they exist and they are not older than x age? 

Currently if your adding lines to the whitelist it takes ~90 seconds for a reload to occur as it re-downloads all lists and generates it from scratch, when your trying to work through what domain is killing something it can be a problem to locate the offending domain If you need to wait 90 seconds between each reload.  

Also stops the hammering of blocklist source sites.

stangri, 2023-02-12:
@dibdot, maybe [this](https://github.com/openwrt/packages/blob/8c8384611b0a8f5d9e6aca46e8cf72918c74198b/net/simple-adblock/files/simple-adblock.init#L948-L1038) will be of use.

dibdot, 2023-02-12:
@mavack that's already implemented. Just Suspend/Resume after your black-/whitelist changes, that re-use your backups and applies the latest local lists.

closed 2023-02-12 by dibdot