openwrt/packages #20485: watchcat: autostart after install

zonyitoo, 2023-02-13:
Maintainer: @kofec @jtkohl 
Environment: Any

Description:

watchcat will auto start after install, because it has set a service reload trigger "watchcat" in `watchcat.init`: https://github.com/openwrt/packages/blob/0876220ba4d2ec3a414328ee0e462e8cc325d039/utils/watchcat/files/watchcat.init#L134-L136

and the `migrate-watchcat` script will commit the `watchcat` config and trigger the service reload: https://github.com/openwrt/packages/blob/0876220ba4d2ec3a414328ee0e462e8cc325d039/utils/watchcat/files/migrate-watchcat#L26

This behavior is unexpected because users didn't do anything to launch watchcat, but it will constantly trigger reboot if the network is not ready.

kofec, 2023-02-14:
Hi I don't fell as maintainer. The maintainer is "Roger D <rogerdammit@gmail.com>".
But if you feel something is wrong, you can always suggest changes and create pull request

zonyitoo, 2023-02-14:
Alright, maintainer changed to @roger- .

I can create a PR for this, but I want to know why it was originally designed like this.