openwrt/packages #20489: tang: bump to v12

rugk, 2023-02-13:
Maintainer: @Tiboris @neheb @cotequeiroz

Description:
Tang has a new v12 release: https://github.com/latchset/tang/releases/tag/v12

Can we bump that?

Interesting part IMHO:
> tang: allow running standalone (https://github.com/latchset/tang/commit/3d6efb1010406ded8f54946de956dc55eb468121)

This is new, and may be useful for OpenWRT.

See also https://github.com/openwrt/packages/issues/15592

rugk, 2023-05-09:
By now v13 is out: https://github.com/latchset/tang/releases/tag/v13

diizzyy, 2023-07-30:
Maintainers get notified by the uscan service, no need to keep it open here
https://sdwalker.github.io/uscan/index.html

closed 2023-07-30 by diizzyy

rugk, 2023-07-31:
Ah I did not know that. But it is still not updated, is it?