openwrt/packages #20496: Wifidog: Last release not  working v1.3.0-8

henok400, 2023-02-14:
Openwrt current OpenWrt 22.03.3 released January 2023 and wifidog v1.3.0-8 released
but wifidog use iptable for that not functional well.
I have installed iptablenft then start working  but only a single device after 2nd device connect the service totally none functional.
In my understanding wifidog v1.3.0-8 need upgrade to use `iptables` based firewall3 to the `nftables` based firewall4.
any one can fix this bugs to use openwrt routers service of captive portal.
wifidog stable openwrt v19.7.10 and less. i build my captive portal based on wifidog that cloud control at cloudwifizone.

predators46, 2023-02-15:
have you tried adding iptables-legacy?

henok400, 2023-02-15:
I have installed iptablenft, iptable legacy then start working but only a single device after 2nd device connect the service totally none functional. This happen above openwrt v19.***. Last stable version of wifidog openwrt 19.***

predators46, 2023-02-15:
does the same thing happen to openwrt 22.02.x?

henok400, 2023-02-15:
I have tested iptables-zz-legacy got the same result wifidog start working then 1 device successfully connect after 2nd device connect problem persist stop working, internet disconnected all device, loop auth. This error happen from openwrt 21.02.x to openwrt 22.03.3. wifidog stable openwrt 19.x.x
here loop this but not connected
```
Wed Feb 15 09:36:50 2023 daemon.info wifidog[1843]: Received connection from 192.168.1.165, spawning worker thread
Wed Feb 15 09:36:50 2023 daemon.info wifidog[1843]: Got client MAC address for ip 192.168.1.165: 1c:bf:ce:22:3e:56
Wed Feb 15 09:36:50 2023 daemon.info wifidog[1843]: Check host detectportal.firefox.com is in whitelist or not
```

predators46, 2023-02-15:
does this not fix?

https://github.com/wifidog/wifidog-gateway/issues/314#issuecomment-1107841121

henok400, 2023-02-15:
I tried that but difficult to compiler. The only way waiting wifidog update version for that new realease wifidog 1.3.0-8 but issue continued.

predators46, 2023-02-16:
maybe you can take a look at this and fix it

https://github.com/wsch32/apfree_wifidog/commits/master/src/fw_iptables.c