openwrt/packages #20498: nft-qos service start error "sh: dynamic: unknown operand"

zhqu1148980644, 2023-02-15:
Maintainer: @rosysong 
Environment: (x86-64, OpenWrt 22.03.3 r20028-43d71ad93e / LuCI openwrt-22.03 branch git-23.035.77007-790b738)

Description:
1. Installed `nft-qos` and `luci-app-nft-qos`, no rate showing in luci, always be `Fetching data`
2. Then I manualy restarted the nft-qos service, and the error `sh: dynamic: unknown operand` shows up.

[image]

[image]

zhqu1148980644, 2023-02-15:
I have another openwrt instance with the same env and the command `nft-qos status/start` shows the same output. However, the rate page works.
If there are any other log files I need to supply?

wehagy, 2025-09-25:
Package removed in the commit https://github.com/openwrt/packages/commit/14bff6e90ef8ebe22d72b4ede398e306ea3b46da.

closed 2025-09-25 by wehagy