openwrt/packages #20509: python3-usb not found - Adding a python library

jimbolaya, 2023-02-16:
Maintainer: @\jefferyto
Environment: x86 22.03.3

Description:

This is kind of a wish list or request to make a package available and I'm not sure if this is the correct place to create it.

I've tagged the Python3 maintainer, but please redirect as appropriate.

I was hoping to find a python3-usb package, similar to many of the general Linux distros like Debian or RedHat. I  searched issues and didn't find any open or closed that mention pyusb, the originating code.

Please let me know if this would be too difficult or if there's a template I could use to add pyusb as a package to OpenWrt. I've modified other standalone OpenWrt packages for my own use (and put them on my own github account), but I'm not sure what's needed for something like this that depends on other packages.

brada4, 2023-03-09:
Like you can jnstall pure  python packages with pip?

jimbolaya, 2023-03-10:
I can try installing packages via pip, though on a router I'd really prefer to keep things cleaner than pip does.

brada4, 2023-03-10:
It is python standard library https://docs.python.org/3/library/index.html turned into separate packages.

jimbolaya, 2023-03-10:
I'm not sure what you're trying to say that it's a "python standard library". 

When installing pip, there's a large list of packages to add, versus likely just a few for pyusb. Reducing the amount of software installed is generally the idea for limited systems is not?

jefferyto, 2023-03-15:
It shouldn't be too hard to add this package - I suggest you take a look at the [Python README](https://github.com/openwrt/packages/blob/56f30520f2413f9f1434def5b533a265912aea1c/lang/python/README.md) and the packages in lang/python for examples (e.g. [python-attrs](https://github.com/openwrt/packages/blob/56f30520f2413f9f1434def5b533a265912aea1c/lang/python/python-attrs/Makefile)).

jimbolaya, 2023-03-15:
OK, thank you for the pointers on where to find what I need.