openwrt/packages #20516: Adblock: OISD rules returning zero domains

pwatk, 2023-02-17:
Maintainer: @dibdot
Environment: 22.03.3 x86_64

Description: OISD rule returning zero domains

Although the recent update (4.1.5-5) fixed the URL's for the OISD block lists, the rules were not updated to account for the change in syntax resulting in zero domains added.

efahl, 2023-02-17:
I noticed this just this morning, too.  The github file names have changed, so an update needs to be made in ``/etc/adblock/adblock.sources.gz``:

Was:
```
https://raw.githubusercontent.com/sjhgvr/oisd/main/dblw_basic.txt
https://raw.githubusercontent.com/sjhgvr/oisd/main/dblw_nsfw.txt
https://raw.githubusercontent.com/sjhgvr/oisd/main/dblw_full.txt
```

Now:
```
https://raw.githubusercontent.com/sjhgvr/oisd/main/domainswild_small.txt
https://raw.githubusercontent.com/sjhgvr/oisd/main/domainswild_nsfw.txt
https://raw.githubusercontent.com/sjhgvr/oisd/main/domainswild_big.txt
```

I can't find the source of record, the stuff in the mirror is grossly different than what I have just re-installed from SNAPSHOT, so I can't make a PR...

efahl, 2023-02-17:
I should also add that I don't think the regex rules have changed, ``adblock.sh reload`` seems to work just fine with only the file name changes.

pwatk, 2023-02-17:
Is it worth switching to the Adblock style lists and adjusting the rules to account for this encase the maintainer chooses to make this change again in the future?

efahl, 2023-02-18:
I don't think so, the file format hasn't changed (ever by my recollection), this was merely a renaming/reorganizing of the files and directory structure.  There was a post on oisd.nl for a while describing the file and directory renaming, but it appears to be gone now.  The only change in file formats on oisd was when dnsmasq 2.86 came out, and he added the dnsmasq2 ``local=`` format files (without changing the old ``server=`` ones), and ``adblock`` doesn't use those anyhow.

pwatk, 2023-02-18:
The maintainer had dropped support for all lists except the Adblock style lists.

He has since reversed this but changed all the file names.

I’m not sure this this is a temporary measure so everyone can transition to the single format or a permanent change due to pressure from the community.

efahl, 2023-02-18:
> The maintainer had dropped support for all lists except the Adblock style lists.

Oh, I had not seen that!  Then, yes, I agree it's probably a good idea to migrate over to that format.  Do you know if they are the domain format, or do they enumerate all possible hosts?  It would be nice if he provided at least a "raw" format domain list...  That's basically what we need for this particular application.

closed 2023-02-18 by dibdot

pwatk, 2023-02-18:
Both lists are the same, only the formatting is different.

punkdthrice, 2023-02-19:
Still not working on 4.1.5-6 ..

(url: https://raw.githubusercontent.com/sjhgvr/oisd/main/dblw_small.txt) returns 404 error when downloading blocklists. 

The maintainer doesn't use 'dblw' anymore.

pwatk, 2023-02-19:
The new [adblock.sources](https://raw.githubusercontent.com/openwrt/packages/master/net/adblock/files/adblock.sources) doesn't use that URL.

A short term solution would be to download the new file...

`curl https://raw.githubusercontent.com/openwrt/packages/master/net/adblock/files/adblock.sources -o /etc/config/dblock.sources`

and reference it in [/etc/config/adblock](https://github.com/openwrt/packages/blob/master/net/adblock/files/README.md)

```
config adblock 'global'
	option adb_srcfile ‘/etc/config/adblock.sources’
```

commit 0b222b7e (2023-02-18): adblock: update 4.1.5-6

* adapted changed oisd downloads (again), fixed #20516
commit fc70290d (2023-02-18): adblock: update 4.1.5-6

* adapted changed oisd downloads (again), fixed #20516

(cherry picked from commit 0b222b7e10ebc8a775d603808e72a7d70ac90972)