openwrt/packages #20568: acme: deprecate the CLI

hgl, 2023-02-25:
Maintainer: @tohojo 

Description:

It seems the convention is to stick util-like commands to the init script. What do you think if we deprecate the CLI and use `/etc/init.d/acme get` instead? Could we simply dropped the CLI, since it's not really an API and the new version of the acme package is only available in snapshot?

tohojo, 2023-02-25:
I think it's fine to just drop it. We should probably support "start" as an argument for the init script, since that's what other scripts do. We could support 'get' in addition, I suppose?

hgl, 2023-02-26:
Do you mean make "start" do what "get" currently does? I like the idea, however:

1. procd seems to [do something special](https://github.com/openwrt/openwrt/blob/888b207f1a6cde4c954e1b3b34f8c2ba60b14cd1/package/base-files/files/etc/rc.common#L142) for the "start" command. Not sure if it's compatible with our one-shot usage?
2. If a user accidentally run `/etc/init.d/acme enable`, does that mean acme will get certificates at every boot (in additional to what cron already does) afterwards?

stangri, 2023-02-26:
I'm actually using the `/usr/lib/acme/acme.sh` provided by the package manually and would be very sad for it to go if I needed to transition to uci config now.

hgl, 2023-02-26:
@stangri We are talking about `/usr/bin/acme`. And you are probably using the previous version of acme. The current version doesn't have `/usr/lib/acme/acme.sh`

tohojo, 2023-02-26:
> The current version doesn't have /usr/lib/acme/acme.sh

I believe what we previously installed in `/usr/lib/acme` was just the upstream `acme.sh`? Which has just moved into `/usr/lib/acme/client` in the new version...

hgl, 2023-02-26:
Yes, @tohojo is right. I haven't touched acme's source code for a while, memory gets rusty. Thanks for the clarification.

Anyway, any comments on the start vs get issue I previously mentioned?

tohojo, 2023-02-26:
I don't think procd has any problems with oneshot services. If I'm reading that rc.common correctly, we can just remove the `START=80` from the init script to make `enable` do nothing. Or we could define a separate `boot()` function to do something else on boot.

I'm not sure it's actually a problem to run a `get` action on boot, though? There's still a timer check for renew actions, so unless the certificate is close to expiring it should be a noop? And if the cert did expire (for instance if the device has been turned off for a while), renewing as soon as possible is probably a good idea. The only drawback I can see is if there's not enough entropy for crypto operations during boot, but I guess that should mostly be a problem for the initial key generation, which should be performed on initial config, not boot...

hgl, 2023-02-28:
I tried the changes on my local device, and opkg started the acme service after I install the package.

@tohojo Although procd can handle oneshot just fine, other tools seem to assume it's for starting a daemon. I think LuCI might also auto-start an init script. This might or might not be intended by a user.

We did make `get` idempotent, but allowing the OS to run "get" in an arbitrary way doesn't seem right. A user might have removed the cron entry and renew in a more controlled way for example.

PR #20581 (2023-03-01): acme: merge cli into init script
Maintainer: @tohojo 
Compile tested: x86 22.03.2
Run tested: x86 22.03.2

Fixes #20568
closed 2023-03-01 by tohojo