openwrt/packages #20576: iptables: iptables 1.8.8 string extension critical bug

levshutov, 2023-02-27:
Maintainer: @PolynomialDivision
Environment: OpenWrt SNAPSHOT r22149-c9df2d5c64

Description:

iptables 1.8.8 release introduced bug in string extension.

It was added here
https://git.netfilter.org/iptables/commit/?id=68ed965b35cdc7b55d4ebc0ba37c1ac078ccbafb

It was fixed here
https://git.netfilter.org/iptables/commit/?id=da5b32fb4656ab69fe1156eb7e36c7c961839e8a

Fix is included in 1.8.9 release.

Demonstration:

Adding string rule
`iptables -A INPUT -m string --string "teststring" --algo bm -j DROP`

Checheking what we have
`iptables -S | grep teststring`
```
-A INPUT -m string --string "teststring" --algo bm --to 65535 -j DROP
```

Trying to delete it
`iptables -D INPUT -m string --string "teststring" --algo bm --to 65535 -j DROP`
```
iptables: Bad rule (does a matching rule exist in that chain?).
```

PolynomialDivision, 2023-02-27:
Thanks. So we should probably update to 1.8.9 soon as possible.

PolynomialDivision, 2023-03-01:
Can you test https://github.com/openwrt/openwrt/pull/12106 ?

BKPepe, 2023-03-02:
This issue is reported to the wrong repository and since I can not transfer it to the main repository (as referenced), where the iptables is located, I'm closing this.

closed 2023-03-02 by BKPepe