openwrt/packages #20582: mwan3 nftset instead of ipset for rules

Suxsem, 2023-03-01:
Dear @feckert,
I'm using openwrt 22.03.03 with mwan3 2.11.4-1.
I'm using firewall4 with nftables and removed any legacy iptables related packages.

The problem is that I'm trying to create some rules based on hostnames instead of IPs.
So I have created a new nftset and told dnsmasq to populate it based on selected hostnames.

How can I use the nftset in the rule? I tried to put the nftset name into the "ipset" field without any luck.

Thank you

dave14305, 2023-03-02:
22.03 still has dnsmasq 2.86, which doesn’t yet support nftsets. But it is supported now in master/snapshots which uses dnsmasq 2.89.

Suxsem, 2023-03-02:
@dave14305 thank you for your reply. Anyway I installed dnsmasq-full and I can see that the nftset is correctly populated with the IP address associated with the desired hostname (nft list ruleset shows them).
The problem is that if I try to put the nftset name in the Mwan3 configuration no rule is shown in the Mwan3 status page and traffic is not correctly routed

feckert, 2023-03-02:
I have not yet ported mwan3 to nft. Therefore, the whole thing only works via the compatibility layer iptables-nft. As far as I can remember, this does not apply to ipsets.
I haven't looked into whether nftables-sets can be accessed in the compatibility layer.

Suxsem, 2023-03-02:
@feckert thank you for looking into this! I don't think iptables-nft can accept an nftset... Maybe there is a way to create an ipset starting from nftset and pass it to the --match-set argument?

feckert, 2023-03-02:
That could work, but I have never done it. Unfortunately, I don't have the time to verify it. You can have a look at it and then report back if it works?

Suxsem, 2023-03-02:
I only found the ipset-translate utility that converts FROM ipset to nftset, not the other way around...

Suxsem, 2023-03-02:
That said I think this must be addressed in some way or another because ipset support has been recently removed from openwrt dnsmasq so expects complaints about it to increase soon :) (https://forum.openwrt.org/t/dnsmasq-full-ipset-support-removed-in-master/150274)

feckert, 2023-05-03:
This issue would be fixed if mwan3 uses native nft in the future (hopfully). But up to now nft rules are not supported

closed 2023-05-03 by feckert