openwrt/packages #20589: samba4: failing to compile on arm_cortex-a9_vfpv3-d16

PalebloodSky, 2023-03-04:
Samba4 began to fail compile for cortex-a9_vfpv3-d16 within roughly the last week.

Output logs are here, use ctrl+F for "a9+vfpv3-d16", errors appear in all files:
https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a9_vfpv3-d16/packages/samba4/

This just started happening within the last week or so and don't think it's GCC 12 because my last build from 10 days ago used GCC 12.2 and it worked great. Sadly this is the only target I use Samba daily for on my WRT32X.

Please fix compiling.

(As an aside, it might be a good time to bump the Samba4 version, 4.17.5 has CVE fixes listed on their website, and Samba 4.18 says to have 3x performance improvement in path based ops.)

PalebloodSky, 2023-03-04:
Dirty work-around: a hopefully short term solution is by installing samba4 from the 22.03.3 release. (The irony is that this wasn't even released for mvebu but the packages carry over.)

In /etc/opkg.conf, commented out the signature check.
In /etc/opkg/customfeeds.conf added:
src/gz openwrt_packages2 https://downloads.openwrt.org/releases/22.03.3/packages/arm_cortex-a9_vfpv3-d16/packages/
src/gz openwrt_base2 https://downloads.openwrt.org/releases/22.03.3/packages/arm_cortex-a9_vfpv3-d16/base/

Then used opkg to install luci-app-samba4, server, etc. from 22.03 feed. Ok it's an older version filled with CVEs compared to Samba 4.17.x in master branch, but at least it installs and works.

1715173329, 2023-03-06:
looks like the package builds again

PalebloodSky, 2023-03-06:
Ah you're right. Not sure how it fixed, I'll check if builds again tomorrow to make sure it's not a fluke then close this issue.

PalebloodSky, 2023-03-06:
Well after about a week of this not compiling it's been compiling again for last 2 days. I'll go ahead and close this.

closed 2023-03-06 by PalebloodSky