openwrt/packages #20596: openblas: compiling issue

trippleflux, 2023-03-05:
Maintainer: @commodo
Environment: x86_64 , snapshot

Description:

Looks like the recent OpenWrt 5a1eb3bb2c27ede3046bb2977008c74a04681db9 commit causing this openblas compiling issues : 
```
Y=x86_64-openwrt-linux-gnu-objcopy OBJDUMP=x86_64-openwrt-linux-gnu-objdump SIZE=x86_64-openwrt-linux-gnu-size CROSS="x86_64-openwrt-linux-gnu-" ARCH="x86_64" CROSS=1 HOSTCC=gcc CROSS_SUFFIX=x86_64-openwrt-linux-gnu- BINARY=64 MAKE_NB_JOBS=-1 NUM_THREADS=2 PREFIX=/usr COMMON_OPT="" TARGET=GENERIC NOFORTRAN=1 ;
make[3]: Entering directory '/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/OpenBLAS-0.3.21'
ccache: invalid option -- 'm'
C Compiler (ccache -m64) is something wrong.
ccache: invalid option -- 'm'
C Compiler (ccache -m64) is something wrong.
/usr/bin/ld: warning: /home/username/works/openwrt/tmp/ccDwe54L.o: missing .note.GNU-stack section implies executable stack
/usr/bin/ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker
ccache: invalid option -- 'm'
C Compiler (ccache -m64) is something wrong.
make[3]: *** [Makefile.prebuild:63: config.h] Error 1
Makefile.system:284: Makefile.conf: No such file or directory
make[3]: *** No rule to make target 'Makefile.conf'.  Stop.
make[3]: Leaving directory '/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/OpenBLAS-0.3.21'
make[2]: *** [Makefile:108: /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/OpenBLAS-0.3.21/.built] Error 2
make[2]: Leaving directory '/home/username/works/openwrt/feeds/packages/libs/openblas'
time: package/feeds/packages/openblas/compile#0.72#1.06#1.67
    ERROR: package/feeds/packages/openblas failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/openblas/compile] Error 1
make[1]: Leaving directory '/home/username/works/openwrt'
make: *** [/home/username/works/openwrt/include/toplevel.mk:231: package/openblas/compile] Error 2
```

brada4, 2023-03-05:
https://github.com/openwrt/packages/blob/a79df49a5fb7fed5a525c72824c966261cbef62e/libs/openblas/Makefile#L79
How did you manage to use ccache?

trippleflux, 2023-03-06:
Already tried to changing HOSTCC into either TARGET_CC & HOSTCC doesn't help

brada4, 2023-03-06:
Your `gcc` command somehow gets using ccache.

trippleflux, 2023-03-06:
Don't know because the openblas OpenWrt Makefile is 'weird' for me as in the openblas package there are cmakelist.txt or using cmake, but on OpenWrt Makefile is not using it.

Also tried the following : 
```
HOSTCC="ccache gcc"
HOSTCC=ccache
```

Currently trying to recompile with OpenWrt 5a1eb3bb2c27ede3046bb2977008c74a04681db9 commit reverted.

brada4, 2023-03-06:
Being small time contributor to OpenBLAS I can tell for certain that you absolutely should not use ccache.
Single source file is supposed to emit different binary objects depending on -D/-U variables like whether input is meant to be single/double/complex or output being minimum or maximum, where ccache will just drop cached copy and make broken resulting library.
Please sanity-check your environment like CC variables, or /usr/bin/gcc being a link to ccache etc.

trippleflux, 2023-03-06:
I am not a OpenBLAS user but in my OpenWrt config it was being selected as module/m.  Didn't realize that, I thought that it was forced selected by another OpenWrt package.
Hmm, getting the same result with reverted OpenWrt 5a1eb3bb2c27ede3046bb2977008c74a04681db9 commit.  Already tried to starting new ssh connection into my build VM with only CFLAGS and FORCE_UNSAFE_CONFIGURE=1 variable, tried to nuke the openwrt build dir and starting from scratch but both giving the same error result.

Should I close this issue?

brada4, 2023-03-06:
Better wait for @commodo - sort of ipk builds perfectly fine all the time. in the meantime post exactly variables you override and edit OP to have complete output of "failed build", it misses "CC"

trippleflux, 2023-03-06:
@commodo 
Thanks, recent OpenWrt commits perhaps broke compilation, especially for x86_64 glibc, my build VM environment doesn't changed but somehow it's hard to do a successful compile currently.

I am currently eyeing at recent OpenWrt's binuntils update into version 2.40

commodo, 2023-03-09:
should be fixed with   https://github.com/openwrt/packages/pull/20624

it took me a while to get that .config building

trippleflux, 2023-03-09:
@commodo 
Thanks a lot, testing it right now

closed 2023-03-09 by neheb