openwrt/packages #20604: gensio: Build error

jefferyto, 2023-03-07:
Maintainer: @WereCatf (gensio), @nxhack (swig)
Environment: armvirt-32, snapshot sdk 2023-03-05

Description:
I am seeing this error when trying to build gensio:

```
make[8]: Entering directory '/media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gensio-2.4.2/c++/swig/pygensio'
/media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32/staging_dir/hostpkg/bin/swig -DHAVE_CONFIG_H -Wall -python -py3 \
        -o pygensio_wrap.cc -c++ \
        -I../../../c++/swig/pygensio -I../../../c++/include \
        -I../../../include \
        -I../../../include -I../../../c++/swig/include \
        ../../../c++/swig/pygensio/pygensio.i
Deprecated command line option: -py3. Ignored, this option is no longer supported.
SWIG:19: Error: PCRE regex matching is not available in this SWIG build.
make[8]: *** [Makefile:852: pygensio_wrap.cc] Error 1
make[8]: Leaving directory '/media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gensio-2.4.2/c++/swig/pygensio'
make[7]: *** [Makefile:569: all-recursive] Error 1
make[7]: Leaving directory '/media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gensio-2.4.2/c++/swig/pygensio'
make[6]: *** [Makefile:427: all-recursive] Error 1
make[6]: Leaving directory '/media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gensio-2.4.2/c++/swig'
make[5]: *** [Makefile:427: all-recursive] Error 1
make[5]: Leaving directory '/media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gensio-2.4.2/c++'
make[4]: *** [Makefile:484: all-recursive] Error 1
make[4]: Leaving directory '/media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gensio-2.4.2'
make[3]: *** [Makefile:416: all] Error 2
make[3]: Leaving directory '/media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gensio-2.4.2'
make[2]: *** [Makefile:207: /media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gensio-2.4.2/.built] Error 2
make[2]: Leaving directory '/home/jeff/Code/personal/openwrt/packages/master/net/gensio'
time: package/feeds/packages/gensio/compile#0.71#0.35#1.88
    ERROR: package/feeds/packages/gensio failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/gensio/compile] Error 1
make[1]: Leaving directory '/media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32'
make: *** [/media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32/include/toplevel.mk:224: package/gensio/compile] Error 2
```

The error message (`PCRE regex matching is not available in this SWIG build`) seems to indicate an issue with swig, though I have two questions:

* swig was configured to build without pcre in 2015 (https://github.com/openwrt/packages/commit/010cb4848d4014d745429db620ab237baf9da8f0) - is this still relevant now?
* gensio appears to build without error on buildbots. It appears that buildbots now have swig installed (https://github.com/openwrt/openwrt/pull/11666#issuecomment-1368998351, https://gitlab.com/openwrt/buildbot/-/commit/3776a9157532e69478ef03236d22254267a4f12e). Is gensio using the swig from the build host instead of the host package here? Is a swig package here still necessary?

nxhack, 2023-03-09:
Perhaps the swig package is configured this way because pcre2 is not required in the host build environment. 

We may need "--with-swig=no" on the gensio package side.

mhei, 2023-03-09:
swig is required to build the python3-gensio binding.

jefferyto, 2023-03-13:
> Perhaps the swig package is configured this way because pcre2 is not required in the host build environment.

There is a pcre2 target package [here](https://github.com/openwrt/packages/tree/master/libs/pcre2), perhaps a host build can be added then swig can use it?

commit 11756e06 (2023-03-22): Merge pull request #20664 from mhei/fix-gensio-build-20604

gensio: fix python bindings build by using a pcre enabled host swig (fixes #20604)
closed 2023-03-22 by mhei