openwrt/packages #20618: icu: build error with ccache

M95D, 2023-03-09:
Maintainer: @nxhack 
Environment: mvebu, Linksys WRT-1900AC v1, host build machine is Gentoo Linux amd64 multilib.

Description:
If ccache is enabled, icu build fails with this error:
```
checking whether to build release libraries... yes
checking for arm-openwrt-linux-clang... ccache
checking whether the C compiler works... no
configure: error: in `/build/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/icu4c-72.1':
configure: error: C compiler cannot create executables
```

config.log errors:
```
configure:3053: $? = 0
configure:3042: ccache -qversion >&5
ccache: invalid option -- 'q'
configure:3053: $? = 1
configure:3073: checking whether the C compiler works
configure:3095: ccache   -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -mfloat-abi=hard -fmacro-prefix-map=/build/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/icu4c-72.1=icu4c-72.1 -Wformat -Werror=format-security -fst
ack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/build/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.2.0_musl_eabi/usr/include -I/build/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.2.0_musl_eabi
/include/fortify -I/build/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.2.0_musl_eabi/include  -L/build/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-12.2.0_musl_eabi/usr/lib -L/build/openwrt/staging_dir/toolcha
in-arm_cortex-a9+vfpv3-d16_gcc-12.2.0_musl_eabi/lib -znow -zrelro  conftest.c  >&5
ccache: invalid option -- 'O'
```

Thanks.

trippleflux, 2023-03-09:
In glibc x86_64 I am getting different error : 
```
mkdir -p /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/icu4c-72.1/data/out
touch /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/icu4c-72.1/.prepared_a9314a683f0d320e712c64f1c9c01935_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/icu4c-72.1/.configured_*
rm -f /home/username/works/openwrt/staging_dir/target-x86_64_glibc_custom/stamp/.icu_installed
(cd /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/icu4c-72.1/./; if [ -x ./runConfigureICU ]; then find /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/icu4c-72.1/ -name config.guess | xargs -r chmod u+w; find /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/icu4c-72.1/ -name config.guess | xargs -r -n1 cp --remove-destination /home/username/works/openwrt/scripts/config.guess; find /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/icu4c-72.1/ -name config.sub | xargs -r chmod u+w; find /home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/icu4c-72.1/ -name config.sub | xargs -r -n1 cp --remove-destination /home/username/works/openwrt/scripts/config.sub; AR="x86_64-openwrt-linux-gnu-gcc-ar" AS="ccache x86_64-openwrt-linux-gnu-gcc -c -mtune=tigerlake -march=tigerlake -O3 -funsafe-math-optimizations -ftree-vectorize -fno-caller-saves -fno-plt -mtune=tigerlake -march=tigerlake -O3 -funsafe-math-optimizations -ftree-vectorize -fhonour-copts -fmacro-prefix-map=/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/icu4c-72.1=icu4c-72.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-gnu-ld NM="x86_64-openwrt-linux-gnu-gcc-nm" CC="ccache x86_64-openwrt-linux-gnu-gcc" GCC="ccache x86_64-openwrt-linux-gnu-gcc" CXX="ccache x86_64-openwrt-linux-gnu-g++" RANLIB="x86_64-openwrt-linux-gnu-gcc-ranlib" STRIP=x86_64-openwrt-linux-gnu-strip OBJCOPY=x86_64-openwrt-linux-gnu-objcopy OBJDUMP=x86_64-openwrt-linux-gnu-objdump SIZE=x86_64-openwrt-linux-gnu-size CFLAGS="-mtune=tigerlake -march=tigerlake -O3 -funsafe-math-optimizations -ftree-vectorize -fno-caller-saves -fno-plt -mtune=tigerlake -march=tigerlake -O3 -funsafe-math-optimizations -ftree-vectorize -fhonour-copts -fmacro-prefix-map=/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/icu4c-72.1=icu4c-72.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-mtune=tigerlake -march=tigerlake -O3 -funsafe-math-optimizations -ftree-vectorize -fno-caller-saves -fno-plt -mtune=tigerlake -march=tigerlake -O3 -funsafe-math-optimizations -ftree-vectorize -fhonour-copts -fmacro-prefix-map=/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/icu4c-72.1=icu4c-72.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_glibc/usr/include -I/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_glibc/include " LDFLAGS="-L/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_glibc/usr/lib -L/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_glibc/lib -znow -zrelro "   ./runConfigureICU Linux/gcc CC="ccache x86_64-openwrt-linux-gnu-gcc" CXX="ccache x86_64-openwrt-linux-gnu-g++" --target=x86_64-openwrt-linux --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --disable-debug --enable-release --enable-shared --enable-static --enable-draft --enable-renaming --disable-tracing --disable-extras --enable-dyload --with-data-packaging=archive --disable-tests --disable-samples --with-cross-build="/home/username/works/openwrt/staging_dir/hostpkg/share/icu/72.1" --prefix=/usr ; fi; )
export CPP= CC=gcc CXX=g++ CPPFLAGS=-I/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_glibc/usr/include -I/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_glibc/include  CFLAGS=  -mtune=tigerlake -march=tigerlake -O3 -funsafe-math-optimizations -ftree-vectorize -fno-caller-saves -fno-plt -mtune=tigerlake -march=tigerlake -O3 -funsafe-math-optimizations -ftree-vectorize -fhonour-copts -fmacro-prefix-map=/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/icu4c-72.1=icu4c-72.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  CXXFLAGS=  -mtune=tigerlake -march=tigerlake -O3 -funsafe-math-optimizations -ftree-vectorize -fno-caller-saves -fno-plt -mtune=tigerlake -march=tigerlake -O3 -funsafe-math-optimizations -ftree-vectorize -fhonour-copts -fmacro-prefix-map=/home/username/works/openwrt/build_dir/target-x86_64_glibc_custom/icu4c-72.1=icu4c-72.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  LDFLAGS=-L/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_glibc/usr/lib -L/home/username/works/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_glibc/lib -znow -zrelro  MAKE=
Running ./configure  CC=ccache x86_64-openwrt-linux-gnu-gcc CXX=ccache x86_64-openwrt-linux-gnu-g++ --target=x86_64-openwrt-linux --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --disable-debug --enable-release --enable-shared --enable-static --enable-draft --enable-renaming --disable-tracing --disable-extras --enable-dyload --with-data-packaging=archive --disable-tests --disable [... cut ...]
    ERROR: package/feeds/packages/icu failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/icu/compile] Error 1
make[1]: Leaving directory '/home/username/works/openwrt'
make: *** [/home/username/works/openwrt/include/toplevel.mk:231: package/icu/compile] Error 2
```

Based on @commodo works on compiling OpenBLAS with ccache, the following edit into icu OpenWrt Makefile works for me : 
```
CONFIGURE_ARGS:= \
	Linux/gcc \
	CC="$(TARGET_CC_NOCACHE)" \
	CXX="$(TARGET_CXX_NOCACHE)" \
```

nxhack, 2023-03-10:
Is this an effect of this?

https://github.com/openwrt/openwrt/commit/5a1eb3bb2c27ede3046bb2977008c74a04681db9

I'm a little busy right now and will check later.

trippleflux, 2023-03-10:
@nxhack 
I was already trying to revert that commit but was still getting previously this [issue](https://github.com/openwrt/packages/issues/20596). But not really sure as I was having not enough rest before.

commodo, 2023-03-10:
> Is this an effect of this?
> 
> [openwrt/openwrt@5a1eb3b](https://github.com/openwrt/openwrt/commit/5a1eb3bb2c27ede3046bb2977008c74a04681db9)
> 
> I'm a little busy right now and will check later.

it's likely that reverting the patch above will fix icu as well;
it also fixes OpenBLAS;
the discussion becomes whether we should revert that patch or fix all upstream makefiles that are affected by it

M95D, 2023-03-10:
It would be simple to revert and ccache will work for those packages too.
Writing patches means more work every time upstream makes changes.

trippleflux, 2023-03-10:
I think the effects is pretty minimal, I am building around ~706 OpenWrt packages for x86_64, and so far only found OpenBLAS and ICU are the problematic one with ccache. As the probable solutions is found, why needs to revert all of those.

brada4, 2023-03-11:
@trippleflux this patch introduced different issue that often predictable or detectable miscompilation with ccache, i.e setting CC=ccache fails to produce executables via `$CC -c -o random test.c` , so many packages of similar GNU autotools versions as ICU would bail same way. Particular error does not mean anything in regard whether ccache will mis-compile later.

trippleflux, 2023-03-11:
Which patch are you referring into? the one from @1715173329 patch above or the one that have been merged into [OpenWrt](https://github.com/openwrt/openwrt/commit/5a1eb3bb2c27ede3046bb2977008c74a04681db9)?

brada4, 2023-03-11:
This one made $CC unable to produce executables, it is fixed:

> [openwrt/openwrt@5a1eb3b](https://github.com/openwrt/openwrt/commit/5a1eb3bb2c27ede3046bb2977008c74a04681db9)
>

closed 2023-03-18 by M95D