openwrt/packages #20663: strongswan: pools support for swanctl?

hgl, 2023-03-14:
Maintainer: @pprindeville @Thermi 
Environment: (put here arch, model, OpenWrt version)

Description:

It seems the swanctl init script doesn't support pools section. Maybe I missed something, but allowing a responder to allocate IP addresses and offer DNS etc seems like a pretty critical feature. I wonder if there is another option that can provide that feature. I don't understand why the strongswan packages have existed for so long without it.

If it turns out allocating IP addresses and offering DNS isn't a commonly used feature. I'd be happy to send a PR that adds support for pools. Let me know if you would accept such a PR.

ptpt52, 2023-03-22:
mark

Thermi, 2023-03-28:
It's commonly used. So we should just merge the code for it once we have it.
In regards to the proxy-arp, one can just use the `farp` plugin. For IPv6, one needs to set up neighbor proxying rules using `ip neigh` (and maybe also enable proxy ndp in `sysctl` for the interface, but I do not know if the latter does the proxying then for all the other local or routed networks too).

hgl, 2023-03-30:
@pprindeville I wonder how do you allocate in-tunnel IPs if not through VIPs? Is it via local_ts?

For Windows and Mac native IKEv2 clients, it seems using VIP is the only way since they don't provide a mean to specify the  in-tunnel IP.

Thermi, 2023-04-08:
No, you can only allocate via pools, not via local_ts or so.

hgl, 2023-04-08:
> No, you can only allocate via pools, not via local_ts or so.

That's my question, I wonder how @pprindeville and others can use strongswan without pools support? Maybe they only use strongswan as an initiator and not a responder?

Thermi, 2023-04-08:
Site-To-Site tunnels.

PR #20687 (2023-04-10): strongswan: add support for pools section
Maintainer: @pprindeville @Thermi
Compile tested: 22.03.2 x86-64
Run tested: 22.03.2 x86-64

Description:
Since I got no response, I went ahead and implemented it. This fixes #20663
closed 2023-04-10 by 1715173329