openwrt/packages #20669: gnutls compile error

zcy85611, 2023-03-15:
latest snapshot, compile fail, didn't change anything before compile, any idea what's wrong with it? thanks

```
make[3]: Entering directory '/workdir/openwrt/package/libs/nettle'
mkdir -p /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
echo '8' | cmp -s - /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/libnettle.version || { echo '8' > /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/libnettle.version;  } 
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-mipsel_24kc_musl/nettle-3.8.1/.pkgdir/libnettle/. /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.libnettle_installed
echo "libnettle" >> /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/nettle.default.install
make[3]: Leaving directory '/workdir/openwrt/package/libs/nettle'
time: package/libs/nettle/compile#0.30#0.05#0.34
make[3]: Entering directory '/workdir/openwrt/feeds/packages/libs/gnutls'
rm -f /workdir/openwrt/build_dir/target-mipsel_24kc_musl/gnutls-3.7.8/.built
touch /workdir/openwrt/build_dir/target-mipsel_24kc_musl/gnutls-3.7.8/.built_check
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/gnutls-3.7.8=gnutls-3.7.8 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/gnutls-3.7.8=gnutls-3.7.8 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/include " LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/lib -znow -zrelro " make  -C /workdir/openwrt/build_dir/target-mipsel_24kc_musl/gnutls-3.7.8/. AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/gnutls-3.7.8=gnutls-3.7.8 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mipsel-openwrt-linux-musl-ld NM="mipsel-openwrt-linux-musl-gcc-nm" CC="mipsel-openwrt-linux-musl-gcc" GCC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CROSS="mipsel-openwrt-linux-musl-" ARCH="mipsel" ;
make[4]: Entering directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/gnutls-3.7.8'
 cd . && automake-1.16 --foreign
configure.ac:41: error: required file 'build-aux/compile' not found
configure.ac:41:   'automake --add-missing' can install 'compile'
configure.ac:31: error: required file 'build-aux/missing' not found
configure.ac:31:   'automake --add-missing' can install 'missing'
aminclude_static.am:52: warning: if $(CODE_COVERAGE_BRANCH_COVERAGE: non-POSIX variable name
aminclude_static.am:52: (probably a GNU make extension)
Makefile.am:176:   'aminclude_static.am' included from here
aminclude_static.am:59: warning: if $(CODE_COVERAGE_BRANCH_COVERAGE: non-POSIX variable name
aminclude_static.am:59: (probably a GNU make extension)
Makefile.am:176:   'aminclude_static.am' included from here
aminclude_static.am:63: warning: ':='-style assignments are not portable
Makefile.am:176:   'aminclude_static.am' included from here
aminclude_static.am:64: warning: code_coverage_v_lcov_cap_$(V: non-POSIX recursive variable expansion
Makefile.am:176:   'aminclude_static.am' included from here
aminclude_static.am:65: warning: code_coverage_v_lcov_cap_$(AM_DEFAULT_VERBOSITY: non-POSIX recursive variable expansion
Makefile.am:176:   'aminclude_static.am' included from here
aminclude_static.am:67: warning: code_coverage_v_lcov_ign_$(V: non-POSIX recursive variable expansion
Makefile.am:176:   'aminclude_static.am' included from here
aminclude_static.am:68: warning: code_coverage_v_lcov_ign_$(AM_DEFAULT_VERBOSITY: non-POSIX recursive variable expansion
Makefile.am:176:   'aminclude_static.am' included from here
aminclude_static.am:70: warning: code_coverage_v_genhtml_$(V: non-POSIX recursive variable expansion
Makefile.am:176:   'aminclude_static.am' included from here
aminclude_static.am:71: warning: code_coverage_v_genhtml_$(AM_DEFAULT_VERBOSITY: non-POSIX recursive variable expansion
Makefile.am:176:   'aminclude_static.am' included from here
aminclude_static.am:73: warning: code_coverage_quiet_$(V: non-POSIX recursive variable expansion
Makefile.am:176:   'aminclude_static.am' included from here
aminclude_static.am:74: warning: code_coverage_quiet_$(AM_DEFAULT_VERBOSITY: non-POSIX recursive variable expansion
Makefile.am:176:   'aminclude_static.am' included from here
aminclude_static.am:78: warning: subst -,_,$(subst .,_,$(1: non-POSIX variable name
aminclude_static.am:78: (probably a GNU make extension)
Makefile.am:176:   'aminclude_static.am' included from here
aminclude_static.am:87: warning: addprefix --directory ,$(CODE_COVERAGE_DIRECTORY: non-POSIX variable name
aminclude_static.am:87: (probably a GNU make extension)
Makefile.am:176:   'aminclude_static.am' included from here
aminclude_static.am:87: warning: call code_coverage_sanitize,$(PACKAGE_NAME: non-POSIX variable name
aminclude_static.am:87: (probably a GNU make extension)
Makefile.am:176:   'aminclude_static.am' included from here
aminclude_static.am:88: warning: addprefix --directory ,$(CODE_COVERAGE_DIRECTORY: non-POSIX variable name
aminclude_static.am:88: (probably a GNU make extension)
Makefile.am:176:   'aminclude_static.am' included from here
aminclude_static.am:90: warning: addprefix --prefix ,$(CODE_COVERAGE_DIRECTORY: non-POSIX variable name
aminclude_static.am:90: (probably a GNU make extension)
Makefile.am:176:   'aminclude_static.am' included from here
aminclude_static.am:100: warning: ':='-style assignments are not portable
Makefile.am:176:   'aminclude_static.am' included from here
doc/Makefile.am: error: required file 'build-aux/depcomp' not found
doc/Makefile.am:   'automake --add-missing' can install 'depcomp'
guile/Makefile.am:68: warning: AM_V_GUILEC_$(V: non-POSIX recursive variable expansion
guile/Makefile.am:69: warning: AM_V_GUILEC_$(AM_DEFAULT_VERBOSITY: non-POSIX recursive variable expansion
guile/Makefile.am:83: warning: '%'-style pattern rules are a GNU make extension
guile/src/Makefile.am:117: warning: '%'-style pattern rules are a GNU make extension
make[4]: *** [Makefile:2188: Makefile.in] Error 1
make[3]: *** [Makefile:269: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/gnutls-3.7.8/.built] Error 2
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/gnutls-3.7.8'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/gnutls'
time: package/feeds/packages/gnutls/compile#5.24#0.11#5.32
    ERROR: package/feeds/packages/gnutls failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/gnutls/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.
```

feckert, 2023-03-15:
I update to the latest stable gnuttls 3.8.0 version. And the compilation error is gone.
```
diff --git a/libs/gnutls/Makefile b/libs/gnutls/Makefile
index 211470205..1634e9681 100644
--- a/libs/gnutls/Makefile
+++ b/libs/gnutls/Makefile
@@ -6,13 +6,13 @@
 include $(TOPDIR)/rules.mk

 PKG_NAME:=gnutls
-PKG_VERSION:=3.7.8
-PKG_RELEASE:=$(AUTORELEASE)
+PKG_VERSION:=3.8.0
+PKG_RELEASE:=1
 PKG_USE_MIPS16:=0

 PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
-PKG_SOURCE_URL:=https://www.gnupg.org/ftp/gcrypt/gnutls/v3.7
-PKG_HASH:=c58ad39af0670efe6a8aee5e3a8b2331a1200418b64b7c51977fb396d4617114
+PKG_SOURCE_URL:=https://www.gnupg.org/ftp/gcrypt/gnutls/v3.8
+PKG_HASH:=0ea0d11a1660a1e63f960f157b197abe6d0c8cb3255be24e1fb3815930b9bdc5

 PKG_MAINTAINER:=Nikos Mavrogiannopoulos <nmav@gnutls.org>
 PKG_LICENSE:=LGPL-2.1-or-later
@@ -21,7 +21,7 @@ PKG_CPE_ID:=cpe:/a:gnu:gnutls

 PKG_BUILD_DEPENDS:=gettext-full/host
 PKG_BUILD_PARALLEL:=1
-PKG_FIXUP:=autoreconf gettext-version
+
 PKG_INSTALL:=1
 PKG_LIBTOOL_PATHS:=. lib

@@ -128,7 +128,6 @@ CONFIGURE_ARGS+= \
        --disable-valgrind-tests \
        --disable-ssl2-support \
        --disable-ssl3-support \
-       --enable-local-libopts \
        --without-idn \
        --with-default-trust-store-dir=/etc/ssl/certs/ \
        --with-included-unistring \
```
It does compile, but I can't say if it works yet, because I have a problem with the zabbix-gnutls package.

lucize, 2023-03-15:
3.7.9 works with

```
--- a/configure.ac      2023-03-15 20:22:10.178636474 +0000
+++ b/configure.ac      2023-03-15 20:22:35.495464673 +0000
@@ -354,7 +354,6 @@
 dnl
 dnl check for gtk-doc
 dnl
-GTK_DOC_CHECK([1.14],[--flavour no-tmpl])

 AM_GNU_GETTEXT([external])
 AM_GNU_GETTEXT_VERSION([0.19])

```

neheb, 2023-03-19:
This was fixed: https://github.com/openwrt/openwrt/commit/f08190e743cc92aaca9f280071aad71bbff83b37

closed 2023-03-19 by neheb