openwrt/packages #20674: acme: missing START= after last overhaul

tiagogaspar8, 2023-03-16:
Maintainer: @tohojo 
Environment: all

Description:

Acme is no longer able to be enabled since the last overhaul because of the missing START=x in the init.d script.
Was this intentional or was this a mistake? I can make a pull request for this of course but I just wanted to clear out what should be the number.

Thanks all.

hgl, 2023-03-16:
It’s intentional. acme is run by cron, not on ever boot.

tiagogaspar8, 2023-03-16:
Ok, I understand, at least this ticket is here for future reference.
Thanks.

closed 2023-03-16 by tiagogaspar8

mipopa, 2024-11-17:
As of acme version 4.0.0, the recommendation (according to the init script) is `Option "webroot" is deprecated, please remove it and change your web server's config so it serves ACME challenge requests from /var/run/acme/challenge`.

Let's take the following scenario:

In `/etc/config/uhttpd` there are two config sections: one for servicing Luci on port 443 and the other one for serving acme challenges on port 80.

```
config uhttpd 'admin'
	list listen_https '[0.0.0.0]:443'
	....

config uhttpd 'acme'
	list listen_http '[0.0.0.0]:80'
	option redirect_https '0'
	option home '/var/run/acme/challenge'
	option no_dirlists '1'
	option rfc1918_filter '1'
	option max_requests '3'
	option max_connections '100'
	option script_timeout '60'
	option network_timeout '30'
	option http_keepalive '20'
	option tcp_keepalive '1'
```

If `/etc/init.d/acme` is not set to run at start-up BEFORE `/etc/init.d/uhttpd`, then there is nothing creating the directory `/var/run/acme/challenge` and uhttpd enters a crash loop. This directory sits in tmpfs and needs to be created by something after a device reboot. Having it in tmpfs instead of the overlay is, by the way, a great idea because the acme challenge files are only needed once in a while and should not be written permanently in the storage. Plus, they could accumulate over time and consume precious space.

Therefore, in my case I modified the `/etc/init.d/acme` script to add the `START=49` line (whereas `/etc/init.d/uhttpd` includes `START=50`) and then run `/etc/init.d/acme enable` to make it start during boot.
There is nothing wrong in having acme getting triggered during boot and by cron. If it doesn't have anything to do (e.g. the certificate not getting close to expiration) then it will gracefully stop.

```
acme-acmesh: Running ACME for <FQDN> with validation_method webroot
acme-acmesh: /usr/lib/acme/client/acme.sh --renew --home /etc/acme -d <FQDN>
Renew: '<FQDN>'
Renew to Le_API=https://acme.zerossl.com/v2/DV90
Skip, Next renewal time is: 2025-01-10T16:08:40Z
Add '--force' to force to renew.
```

I would recommend developers to port this change into the main code to make acme start at boot and ensure compatibility with uhttpd if the validation method is webroot.

tohojo, 2024-11-19:
Hmm, right, fair point. I don't think we should run the regular start routine at boot, though. But we could have a `boot()` routine that just creates the directory. Please see #25409 for an implementation of that. @mipopa, care to test if that approach fixes the issue for you? :)

mipopa, 2024-11-20:
More than happy to test. My workaround that worked for me was to add ‘START=49’ (anything lower than uhttpd’s value) in ˋ/etc/init.d/acmeˋ and then enable it. May not be clean but solved for me. So let me know what and when to test.