openwrt/packages #20677: Dovecot -  Dovecot does not build since the commits introduced on 13th March.

tankwrt, 2023-03-16:
Maintainer: Lucian Cristian <lucian.cristian@gmail.com>
Environment: (Mediatek Ralink Mips, MT7621 Board, ZBT-WE1326, model, Openwrt Master)

Description:

Dovecot does not build since the commits introduced on 13th March.

```
make[3]: Entering directory '/mnt/256/master/feeds/packages/mail/dovecot'
touch /mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18/.prepared_de73c9b0e2b8bf7644c7b5ae9dd69c76_6664517399ebbbc92a37c5bb081b5c53_check
. /mnt/256/master/include/shell.sh; /mnt/256/master/staging_dir/host/bin/libdeflate-gzip -dc /mnt/256/master/dl/dovecot-2.3.18.tar.gz | tar -C /mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18

Applying ./patches/001-configure_in.patch using plaintext: 
patching file m4/fd_passing.m4
patching file m4/glibc.m4
patching file m4/ioloop.m4
patching file m4/mmap_write.m4
patching file m4/want_gssapi.m4

Applying ./patches/050-fix-dovecot-config-for-cross-compile.patch using plaintext: 
patching file Makefile.am

Applying ./patches/100-openssl-deprecated.patch using plaintext: 
patching file src/lib-ssl-iostream/dovecot-openssl-common.c
patching file src/lib-ssl-iostream/iostream-openssl-context.c

Applying ./patches/110-openssl-engine.patch using plaintext: 
patching file src/lib-dcrypt/dcrypt-openssl.c
patching file src/lib-ssl-iostream/dovecot-openssl-common.c
touch /mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18/.prepared_de73c9b0e2b8bf7644c7b5ae9dd69c76_6664517399ebbbc92a37c5bb081b5c53
rm -f /mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18/.configured_*
rm -f /mnt/256/master/staging_dir/target-mipsel_24kc_musl/stamp/.dovecot_installed
(cd /mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf ./autom4te.cache; [ -e ./config.rpath ] || ln -s /mnt/256/master/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/mnt/256/master/staging_dir/host/bin/autom4te AUTOCONF=/mnt/256/master/staging_dir/host/bin/autoconf AUTOMAKE=/mnt/256/master/staging_dir/host/bin/automake ACLOCAL=/mnt/256/master/staging_dir/host/bin/aclocal AUTOHEADER=/mnt/256/master/staging_dir/host/bin/autoheader LIBTOOLIZE=/mnt/256/master/staging_dir/host/bin/libtoolize LIBTOOL=/mnt/256/master/staging_dir/host/bin/libtool M4=/mnt/256/master/staging_dir/host/bin/m4 AUTOPOINT=true LIBTOOLIZE='/mnt/256/master/staging_dir/host/bin/libtoolize --install' /mnt/256/master/staging_dir/host/bin/autoreconf -v -f -i -s -B /mnt/256/master/staging_dir/host/share/aclocal -I /mnt/256/master/staging_dir/target-mipsel_24kc_musl/host/share/aclocal -I /mnt/256/master/staging_dir/hostpkg/share/aclocal -I /mnt/256/master/staging_dir/target-mipsel_24kc_musl/usr/share/aclocal -I m4 -I . . || true; fi; );
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: /mnt/256/master/staging_dir/host/bin/aclocal -I /mnt/256/master/staging_dir/target-mipsel_24kc_musl/host/share/aclocal -I /mnt/256/master/staging_dir/hostpkg/share/aclocal -I /mnt/256/master/staging_dir/target-mipsel_24kc_musl/usr/share/aclocal -I m4 -I . --force -I m4
configure.ac:285: warning: macro 'AM_ICONV' not found in library
autoreconf: configure.ac: tracing
autoreconf: running: /mnt/256/master/staging_dir/host/bin/libtoolize --install --force
OpenWrt-libtoolize: putting auxiliary files in '.'.
OpenWrt-libtoolize: linking file './config.guess'
OpenWrt-libtoolize: linking file './config.sub'
OpenWrt-libtoolize: linking file './install-sh'
OpenWrt-libtoolize: linking file './ltmain.sh'
OpenWrt-libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
OpenWrt-libtoolize: linking file 'm4/libtool.m4'
OpenWrt-libtoolize: linking file 'm4/ltoptions.m4'
OpenWrt-libtoolize: linking file 'm4/ltsugar.m4'
OpenWrt-libtoolize: linking file 'm4/ltversion.m4'
OpenWrt-libtoolize: linking file 'm4/lt~obsolete.m4'
autoreconf: configure.ac: not using Intltool
autoreconf: configure.ac: not using Gtkdoc
autoreconf: running: /mnt/256/master/staging_dir/host/bin/aclocal -I /mnt/256/master/staging_dir/target-mipsel_24kc_musl/host/share/aclocal -I /mnt/256/master/staging_dir/hostpkg/share/aclocal -I /mnt/256/master/staging_dir/target-mipsel_24kc_musl/usr/share/aclocal -I m4 -I . --force -I m4
configure.ac:285: warning: macro 'AM_ICONV' not found in library
autoreconf: running: /mnt/256/master/staging_dir/host/bin/autoconf --include=/mnt/256/master/staging_dir/target-mipsel_24kc_musl/host/share/aclocal --include=/mnt/256/master/staging_dir/hostpkg/share/aclocal --include=/mnt/256/master/staging_dir/target-mipsel_24kc_musl/usr/share/aclocal --include=m4 --include=. --prepend-include=/mnt/256/master/staging_dir/host/share/aclocal --force
configure.ac:62: warning: The macro `AC_ERROR' is obsolete.
configure.ac:62: You should run autoupdate.
./lib/autoconf/oldnames.m4:34: AC_ERROR is expanded from...
m4/test_with.m4:2: TEST_WITH is expanded from...
configure.ac:62: the top level
configure.ac:67: warning: The macro `AC_ERROR' is obsolete.
configure.ac:67: You should run autoupdate.
./lib/autoconf/oldnames.m4:34: AC_ERROR is expanded from...
m4/test_with.m4:2: TEST_WITH is expanded from...
configure.ac:67: the top level
configure.ac:72: warning: The macro `AC_ERROR' is obsolete.
configure.ac:72: You should run autoupdate.
./lib/autoconf/oldnames.m4:34: AC_ERROR is expanded from...
[... 731 lines cut ...]
configure.ac:762: the top level
configure.ac:764: warning: The macro `AC_ERROR' is obsolete.
configure.ac:764: You should run autoupdate.
./lib/autoconf/oldnames.m4:34: AC_ERROR is expanded from...
m4/want_icu.m4:1: DOVECOT_WANT_ICU is expanded from...
configure.ac:764: the top level
configure.ac:285: error: possibly undefined macro: AM_ICONV
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: error: /mnt/256/master/staging_dir/host/bin/autoconf failed with exit status: 1
(cd /mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18/./; if [ -x ./configure ]; then find /mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18/ -name config.guess | xargs -r chmod u+w; find /mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18/ -name config.guess | xargs -r -n1 cp --remove-destination /mnt/256/master/scripts/config.guess; find /mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18/ -name config.sub | xargs -r chmod u+w; find /mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18/ -name config.sub | xargs -r -n1 cp --remove-destination /mnt/256/master/scripts/config.sub; AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18=dovecot-2.3.18 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mipsel-openwrt-linux-musl-ld NM="mipsel-openwrt-linux-musl-gcc-nm" CC="mipsel-openwrt-linux-musl-gcc" GCC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18=dovecot-2.3.18 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18=dovecot-2.3.18 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   " CPPFLAGS="-I/mnt/256/master/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/usr/include -I/mnt/256/master/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/include/fortify -I/mnt/256/master/staging_dir/toolchain-mipsel_24kc_gcc-12.2.0_musl/include   " L [... cut ...]
make[3]: Leaving directory '/mnt/256/master/feeds/packages/mail/dovecot'
time: package/feeds/packages/dovecot/compile#11.73#0.81#13.35
    ERROR: package/feeds/packages/dovecot failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/dovecot/compile] Error 1
make[2]: Leaving directory '/mnt/256/master'
make[1]: *** [package/Makefile:110: /mnt/256/master/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/256/master'
make: *** [/mnt/256/master/include/toplevel.mk:231: world] Error 2

```

tankwrt, 2023-03-16:
Hi, unfortunatelly I tried  this morning master build incluiding that patch 
 and also newer https://github.com/openwrt/openwrt/commit/f08190e743cc92aaca9f280071aad71bbff83b37 
and the issue persists

oskarirauta, 2023-03-16:
Similar issues with multiple packages since those changes; and that commit didn't fix the issue.
sqlite3 and open-vm-tools had these issues, and who knows how many others, error message for both of these packages looks like this:
```
libtool: Version mismatch error.  This is libtool 2.4.7, but the
libtool: definition of this LT_INIT comes from libtool 2.4.2.
libtool: You should recreate aclocal.m4 with macros from libtool 2.4.7
libtool: and run autoconf again.
make[6]: *** [Makefile:522: install-libLTLIBRARIES] Error 63
make[6]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl/open-vm-tools-12.1.5-20735119/libguestStoreClient'
```

I tried with setting pkg_fixup to lib tool-patch but unfortunately it wasn't sufficient.

hauke, 2023-03-16:
Some build bot runs are looking fine now: https://downloads.openwrt.org/snapshots/faillogs/arm_arm926ej-s/
I think the other failing ones are still using an old SDK.

For me dovecot is compiling in recent master. Please try to manually delete it first: `make package/dovecot/clean V=99`. The fixes are for the autoreconf script and I think it is not run again in case it was run before even when it failed. 

I used this configuration and did not use the SDK:
````
hauke@hauke-t480:~/openwrt/openwrt$ ./scripts/diffconfig.sh 
CONFIG_TARGET_malta=y
CONFIG_TARGET_malta_be=y
CONFIG_TARGET_malta_be_Default=y
CONFIG_GNUTLS_ALPN=y
CONFIG_GNUTLS_ANON=y
CONFIG_GNUTLS_DTLS_SRTP=y
CONFIG_GNUTLS_HEARTBEAT=y
CONFIG_GNUTLS_OCSP=y
CONFIG_GNUTLS_PSK=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_GNUTLS=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_PROXY=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_dovecot=y
CONFIG_PACKAGE_dovecot-utils=y
CONFIG_PACKAGE_kmod=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libbz2=y
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_libcurl=y
CONFIG_PACKAGE_libgmp=y
CONFIG_PACKAGE_libgnutls=y
CONFIG_PACKAGE_libnettle=y
CONFIG_PACKAGE_libnghttp2=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_zlib=y
hauke@hauke-t480:~/openwrt/openwrt$ ./scripts/getver.sh 
r22281-f08190e743cc
```

oskarirauta, 2023-03-16:
Didn't help. I also removed buildir_dir/target*/dovecot** manually, same result.

```
make[5]: Entering directory '/usr/src/openwrt/build_dir/target-x86_64_musl/dovecot-2.3.18/src'
Making all in lib-test
make[6]: Entering directory '/usr/src/openwrt/build_dir/target-x86_64_musl/dovecot-2.3.18/src/lib-test'
/bin/sh ../../libtool  --tag=CC   --mode=compile x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../..  -I../../src/lib -I../../src/lib-charset -I/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_musl/usr/include -I/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_musl/include/fortify -I/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-12.2.0_musl/include -I/usr/src/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-full/include -I/usr/src/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-full/include   -std=gnu99 -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/usr/src/openwrt/build_dir/target-x86_64_musl/dovecot-2.3.18=dovecot-2.3.18 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/usr/src/openwrt/staging_dir/target-x86_64_musl/usr/lib/libiconv-full/include -I/usr/src/openwrt/staging_dir/target-x86_64_musl/usr/lib/libintl-full/include  -fstack-protector-strong -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=2 -mfunction-return=keep -mindirect-branch=keep -Wall -W -Wmissing-prototypes -Wmissing-declarations -Wpointer-arith -Wchar-subscripts -Wformat=2 -Wbad-function-cast -fno-builtin-strftime -Wstrict-aliasing=2 -I/usr/src/openwrt/staging_dir/target-x86_64_musl/usr/include  -c -o fuzzer.lo fuzzer.c
libtool: Version mismatch error.  This is libtool 2.4.7, but the
libtool: definition of this LT_INIT comes from libtool 2.4.2.
libtool: You should recreate aclocal.m4 with macros from libtool 2.4.7
libtool: and run autoconf again.
make[6]: *** [Makefile:574: fuzzer.lo] Error 63
make[6]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl/dovecot-2.3.18/src/lib-test'
make[5]: *** [Makefile:570: all-recursive] Error 1
make[5]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl/dovecot-2.3.18/src'
make[4]: *** [Makefile:700: all-recursive] Error 1
make[4]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl/dovecot-2.3.18'
make[3]: *** [Makefile:544: all] Error 2
make[3]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl/dovecot-2.3.18'
make[2]: *** [Makefile:138: /usr/src/openwrt/build_dir/target-x86_64_musl/dovecot-2.3.18/.built] Error 2
make[2]: Leaving directory '/usr/src/openwrt/feeds/packages/mail/dovecot'
time: package/feeds/packages/dovecot/compile#24.80#2.14#0.00
    ERROR: package/feeds/packages/dovecot failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/dovecot/compile] Error 1
make[1]: Leaving directory '/usr/src/openwrt'
make: *** [/usr/src/openwrt/include/toplevel.mk:231: package/dovecot/compile] Error 2
```

oskarirauta, 2023-03-16:
I re-built most of my tools, but unfortunately result is the same.

tankwrt, 2023-03-16:
I tried make package/dovecot/clean V=99 and then make -j1 V=cs, but same error:

```
make[3]: Entering directory '/mnt/256/master/feeds/packages/mail/dovecot'
touch /mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18/.prepared_95daebecd6db9911ac2eef06cb725cef_6664517399ebbbc92a37c5bb081b5c53_check
. /mnt/256/master/include/shell.sh; /mnt/256/master/staging_dir/host/bin/libdeflate-gzip -dc /mnt/256/master/dl/dovecot-2.3.18.tar.gz | tar -C /mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18

Applying ./patches/001-configure_in.patch using plaintext: 
patching file m4/fd_passing.m4
patching file m4/glibc.m4
patching file m4/ioloop.m4
patching file m4/mmap_write.m4
patching file m4/want_gssapi.m4

Applying ./patches/050-fix-dovecot-config-for-cross-compile.patch using plaintext: 
patching file Makefile.am

Applying ./patches/100-openssl-deprecated.patch using plaintext: 
patching file src/lib-ssl-iostream/dovecot-openssl-common.c
patching file src/lib-ssl-iostream/iostream-openssl-context.c

Applying ./patches/110-openssl-engine.patch using plaintext: 
patching file src/lib-dcrypt/dcrypt-openssl.c
patching file src/lib-ssl-iostream/dovecot-openssl-common.c
touch /mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18/.prepared_95daebecd6db9911ac2eef06cb725cef_6664517399ebbbc92a37c5bb081b5c53
rm -f /mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18/.configured_*
rm -f /mnt/256/master/staging_dir/target-mipsel_24kc_musl/stamp/.dovecot_installed
(cd /mnt/256/master/build_dir/target-mipsel_24kc_musl/dovecot-2.3.18; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf ./autom4te.cache; [ -e ./config.rpath ] || ln -s /mnt/256/master/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/mnt/256/master/staging_dir/host/bin/autom4te AUTOCONF=/mnt/256/master/staging_dir/host/bin/autoconf AUTOMAKE=/mnt/256/master/staging_dir/host/bin/automake ACLOCAL=/mnt/256/master/staging_dir/host/bin/aclocal AUTOHEADER=/mnt/256/master/staging_dir/host/bin/autoheader LIBTOOLIZE=/mnt/256/master/staging_dir/host/bin/libtoolize LIBTOOL=/mnt/256/master/staging_dir/host/bin/libtool M4=/mnt/256/master/staging_dir/host/bin/m4 AUTOPOINT=true GTKDOCIZE=true LIBTOOLIZE='/mnt/256/master/staging_dir/host/bin/libtoolize --install' /mnt/256/master/staging_dir/host/bin/autoreconf -v -f -i -s -B /mnt/256/master/staging_dir/host/share/aclocal -I /mnt/256/master/staging_dir/target-mipsel_24kc_musl/host/share/aclocal -I /mnt/256/master/staging_dir/hostpkg/share/aclocal -I /mnt/256/master/staging_dir/target-mipsel_24kc_musl/usr/share/aclocal -I m4 -I . . || true; fi; );
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: /mnt/256/master/staging_dir/host/bin/aclocal -I /mnt/256/master/staging_dir/target-mipsel_24kc_musl/host/share/aclocal -I /mnt/256/master/staging_dir/hostpkg/share/aclocal -I /mnt/256/master/staging_dir/target-mipsel_24kc_musl/usr/share/aclocal -I m4 -I . --force -I m4
configure.ac:285: warning: macro 'AM_ICONV' not found in library
autoreconf: configure.ac: tracing
autoreconf: running: /mnt/256/master/staging_dir/host/bin/libtoolize --install --force
OpenWrt-libtoolize: putting auxiliary files in '.'.
OpenWrt-libtoolize: linking file './config.guess'
OpenWrt-libtoolize: linking file './config.sub'
OpenWrt-libtoolize: linking file './install-sh'
OpenWrt-libtoolize: linking file './ltmain.sh'
OpenWrt-libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
OpenWrt-libtoolize: linking file 'm4/libtool.m4'
OpenWrt-libtoolize: linking file 'm4/ltoptions.m4'
OpenWrt-libtoolize: linking file 'm4/ltsugar.m4'
OpenWrt-libtoolize: linking file 'm4/ltversion.m4'
OpenWrt-libtoolize: linking file 'm4/lt~obsolete.m4'
autoreconf: configure.ac: not using Intltool
autoreconf: configure.ac: not using Gtkdoc
autoreconf: running: /mnt/256/master/staging_dir/host/bin/aclocal -I /mnt/256/master/staging_dir/target-mipsel_24kc_musl/host/share/aclocal -I /mnt/256/master/staging_dir/hostpkg/share/aclocal -I /mnt/256/master/staging_dir/target-mipsel_24kc_musl/usr/share/aclocal -I m4 -I . --force -I m4
configure.ac:285: warning: macro 'AM_ICONV' not found in library
autoreconf: running: /mnt/256/master/staging_dir/host/bin/autoconf --include=/mnt/256/master/staging_dir/target-mipsel_24kc_musl/host/share/aclocal --include=/mnt/256/master/staging_dir/hostpkg/share/aclocal --include=/mnt/256/master/staging_dir/target-mipsel_24kc_musl/usr/share/aclocal --include=m4 --include=. --prepend-include=/mnt/256/master/staging_dir/host/share/aclocal --force
configure.ac:62: warning: The macro `AC_ERROR' is obsolete.
configure.ac:62: You should run autoupdate.
./lib/autoconf/oldnames.m4:34: AC_ERROR is expanded from...
m4/test_with.m4:2: TEST_WITH is expanded from...
configure.ac:62: the top level
configure.ac:67: warning: The macro `AC_ERROR' is obsolete.
configure.ac:67: You should run autoupdate.
./lib/autoconf/oldnames.m4:34: AC_ERROR is expanded from...
m4/test_with.m4:2: TEST_WITH is expanded from...
configure.ac:67: the top level
configure.ac:72: warning: The macro `AC_ERROR' is obsolete.
configure.ac:72: You should run autoupdate.
./lib/autoconf/oldnames.m4:34: AC_ERROR is expanded from...
[... cut ...]
make[3]: Leaving directory '/mnt/256/master/feeds/packages/mail/dovecot'
time: package/feeds/packages/dovecot/compile#5.39#0.27#6.60
    ERROR: package/feeds/packages/dovecot failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/dovecot/compile] Error 1
make[2]: Leaving directory '/mnt/256/master'
make[1]: *** [package/Makefile:110: /mnt/256/master/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/256/master'
make: *** [/mnt/256/master/include/toplevel.mk:231: world] Error 2

```

oskarirauta, 2023-03-18:
temporary work-around:
I made open-vm-tools clean and then compile, to get this error..
then I went to open-vm-tools's build dir and executed this:

`cat /usr/src/openwrt/staging_dir/host/share/aclocal/libtool.m4 /usr/src/openwrt/staging_dir/host/share/aclocal/ltoptions.m4 /usr/src/openwrt/staging_dir/host/share/aclocal/ltversion.m4 >> aclocal.m4`

then build open-vm-tools again without cleaning it first, package was built succesfully.

Now I tried to do same for dovecot, but it changed nothing.

oskarirauta, 2023-03-18:
Is there someone who doesn't have these issues or do these need patching?

oskarirauta, 2023-03-19:
I started from fresh by cloning new build tree and building everything from beginning. Build is in progress as we speak, but looks promising as on of culprit packages has this far been able to build without this issue: sqlite3.

lucize, 2023-03-21:
for me sqlite3 worked after make dirclean on Ubuntu 22.04

micmac1, 2023-03-25:
Looking at the provided logs autoreconf fails because the iconv macro is missing (iconv.m4). If I remember correctly the macro was provided earlier by libiconv package, but that's no longer the case.

I fixed the same issue in asterisk-chan-sccp package. Add this to dovecot Makefile:

```
# need iconv.m4, otherwise error during autoreconf
PKG_BUILD_DEPENDS:=gettext-full
```

That should make it work.

tankwrt, 2023-03-26:
> Looking at the provided logs autoreconf fails because the iconv macro is missing (iconv.m4). If I remember correctly the macro was provided earlier by libiconv package, but that's no longer the [...]

Thank you! That have worked, I was able to compile master again with Dovecot 
Should I open a pull request or this is somthing that the maintainer of the package takes care?

lucize, 2023-03-26:
@tankwrt please open and I'll approve it

commit 96145db7 (2023-05-10): dovecot: Fix iconv macro is missing compile error

Fix a Dovecot compile error when building with no other packages than
the default in master build, because iconv macro is missing.

Fixes: #20677
commit 4c8d5853 (2023-05-14): dovecot: Fix iconv macro is missing compile error

Fix a Dovecot compile error when building with no other packages than
the default in master build, because iconv macro is missing.

Fixes: #20677

(cherry picked from commit 96145db78af6afa27d8fa857d2af384b1c623259)
closed 2023-05-10 by 1715173329