openwrt/packages #20680: luci-app-wifischedule - idea - force disable if only specified stations associated

happydashery, 2023-03-16:
Maintainer: @newkit

Environment: (Openwrt 22.03.3, Unifi 6 LR v1, ARMv8 Processor rev 4, mediatek/mt7622)

Description:
May I start by saying I love your work, and this is an idea, not an issue. I would like to use the luci-app-wifischedule to turn off wireless at night. If I'm out there working though, I'd rather it stay on. Therefore, I don't want to force disable. However, I have an Alexa speaker out there which I don't always remember to switch off. Is it possible to add an option of maybe allowing you to input mac addresses of devices that can be ignored if they are still connected, and the wifi switched off according to the schedule?

Thanks

Happy

brada4, 2023-03-17:
Option is available in wifischedule [command line uci ](https://github.com/openwrt/packages/blob/07e032ee1b8372ff56834d6170d67620de3d5d7c/net/wifischedule/net/etc/config/wifi_schedule#L6).

Luci's ideas go to luci repo.

happydashery, 2023-03-20:
> Option is available in wifischedule [command line uci ](https://github.com/openwrt/packages/blob/07e032ee1b8372ff56834d6170d67620de3d5d7c/net/wifischedule/net/etc/config/wifi_schedule#L6).
> 
> Luci's ideas go to luci repo.

Hi Brada4

Apologies, I thought I was posting it in the right place, goes to show how much I know. Thanks for your reply though, that's answered my issue nicely anyways.

Best wishes 

Happy

closed 2023-03-20 by happydashery