openwrt/packages #20681: docker: Build error involving PKG_GIT_SHORT_COMMIT

jefferyto, 2023-03-17:
Maintainer: @G-M0N3Y-2503 
Environment: armvirt-32, 2023-03-13 snapshot sdk

Description:
I'm seeing this build error:

```
make[2]: Entering directory '/home/jeff/Code/personal/openwrt/packages/master/utils/docker'
touch /media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/docker-23.0.1/.prepared_af17b59620e4e56c33d25fa1319b0df9_6664517399ebbbc92a37c5bb081b5c53_check
. /media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32/include/shell.sh; /media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32/staging_dir/host/bin/libdeflate-gzip -dc /media/jeff/Jekyll/Downloads/openwrt/dl/docker-23.0.1.tar.gz | tar -C /media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/docker-23.0.1 --strip-components 1 -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/docker-23.0.1
# Verify PKG_GIT_SHORT_COMMIT
( EXPECTED_PKG_GIT_SHORT_COMMIT=$( /home/jeff/Code/personal/openwrt/packages/master/utils/docker/../dockerd/git-short-commit.sh 'github.com/docker/cli' 'v23.0.1' '/media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32/tmp/git-short-commit/docker-23.0.1' ); if [ "${EXPECTED_PKG_GIT_SHORT_COMMIT}" != "a5ee5b1" ]; then echo "ERROR: Expected 'PKG_GIT_SHORT_COMMIT:=${EXPECTED_PKG_GIT_SHORT_COMMIT}', found 'PKG_GIT_SHORT_COMMIT:=a5ee5b1'"; exit 1; fi )
Trying remote 'github.com/docker/cli'
fatal: 'github.com/docker/cli' does not appear to be a git repository
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
Trying remote 'https://github.com/docker/cli'
remote: Enumerating objects: 3804, done.
remote: Counting objects: 100% (3804/3804), done.
remote: Compressing objects: 100% (3075/3075), done.
remote: Total 3804 (delta 624), reused 1826 (delta 425), pack-reused 0
Receiving objects: 100% (3804/3804), 5.99 MiB | 2.03 MiB/s, done.
Resolving deltas: 100% (624/624), done.
From https://github.com/docker/cli
 * tag                         v23.0.1    -> FETCH_HEAD
HEAD is now at a5ee5b1dfc9b Merge pull request #4018 from thaJeztah/23.0_backport_fix_ci_events
ERROR: Expected 'PKG_GIT_SHORT_COMMIT:=a5ee5b1dfc9b', found 'PKG_GIT_SHORT_COMMIT:=a5ee5b1'
make[2]: *** [Makefile:83: /media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/docker-23.0.1/.prepared_af17b59620e4e56c33d25fa1319b0df9_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[2]: Leaving directory '/home/jeff/Code/personal/openwrt/packages/master/utils/docker'
time: package/feeds/packages/docker/compile#1.34#1.26#8.69
    ERROR: package/feeds/packages/docker failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/docker/compile] Error 1
make[1]: Leaving directory '/media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32'
make: *** [/media/jeff/Jekyll/Downloads/openwrt/testing/armvirt-32/include/toplevel.mk:224: package/docker/compile] Error 2
```

I have `core.abbrev` in my git config set to `12`, this is why the commit hashes do not "match". I suggest testing for the full commit hash instead of the short hash.

jefferyto, 2023-03-17:
Also I get the same error when trying to build dockerd.

G-M0N3Y-2503, 2023-03-17:
The use for PKG_GIT_SHORT_COMMIT is for cosmetic purposes inside the CLI and matches the upstream length.  
Perhaps we could modify `utils/dockerd/git-short-commit.sh` to use `git -c core.abbrev=auto` to override users default for this command.