openwrt/packages #20692: nginx: increase server_names_hash_bucket_size in uci.conf.template

Curious-r, 2023-03-20:
Maintainer: @hgl 
Environment: (put here arch, model, OpenWrt version)
ARMv7 Processor rev 5 (v7l)
OpenWrt SNAPSHOT r0-4bfbecb / LuCI Master git-23.075.21248-51db83c

Description:
On 32-bit devices, the default value of server_names_hash_bucket_size is very low. So much so that when we use a custom domain name, nginx will report an error like this:
```
2023/03/20 07:17:30 [emerg] 17323#0: could not build server_names_hash, you should increase server_names_hash_bucket_size: 32
nginx: configuration file /etc/nginx/uci.conf test failed
```
At this time, the custom domain name is just a regular third-level domain name, like `gateway.example.com`.
We can easily add `server_names_hash_bucket_size: 32` to the http block to solve this problem. But since this problem is so common, can we add it to `uci.conf.temlate`?

hgl, 2023-03-20:
I’m not the maintainer.

Curious-r, 2023-03-20:
> I’m not the maintainer.

Sorry...

Curious-r, 2023-03-20:
> If you comment out parameter - does it detect different hash bucket size from cache line size? It is not about 32bit vs64bit, it is minimum grain of memory access (to the powers of 2)

It may be that my statement is inaccurate. In my device, by default, this parameter is not explicitly declared. At this point, if I add a long domain name such as example.example.com, nginx will report:
```
2023/03/20 07:17:30 [emerg] 17323#0: could not build server_names_hash, you should increase server_names_hash_bucket_size: 32
nginx: configuration file /etc/nginx/uci.conf test failed
```
I'm guessing it's because many 32-bit-cpu's cache line is relatively small, such as cortex-a7.
There is description from nginx.org:

```
If a large number of server names are defined, or unusually long server names are defined, tuning the server_names_hash_max_size and server_names_hash_bucket_size directives at the http level may become necessary. The default value of the server_names_hash_bucket_size directive may be equal to 32, or 64, or another value, depending on CPU cache line size. If the default value is 32 and server name is defined as “too.long.server.name.example.org” then nginx will fail to start and display the error message:

could not build the server_names_hash,
you should increase server_names_hash_bucket_size: 32
```
So I have to explicitly define a larger parameter, then the `uci.conf` can pass the test.

peter-stadler, 2023-03-20:
I think there is no single value that is right for most users.

So, I would recommend that you create a custom file with this paramater, e.g. `/etc/nginx/conf.d/server_names_hash.conf`:
```
server_names_hash_bucket_size 64;
# server_names_hash_max_size 2056;
```
Then the line `include conf.d/*.conf` activates it for the main http config.

Adopting the upstream `uci.conf.template` would be possible, but makes it more difficult for users to change the parameter. They need to edit the local `uci.conf.template` and thereafter follow upstream updates to this file manually.

Curious-r, 2023-03-20:
> I think there is no single value that is right for most users.
>
> So, I would recommend that you create a custom file with this paramater, e.g. `/etc/nginx/conf.d/server_names_hash.conf`:
>
> ``` [...]

Copy that, thanks for getting back to me. 😆

closed 2023-03-20 by Curious-r

Curious-r, 2023-03-21:
> Probably worth setting that parameter to 64 by default to match x86_64 and aarch64 autodetected default that runs half of the internet nowadays... Or at least not force anything.

Yes, I feel the same way. 64 is a very suitable default value. 🙌