openwrt/packages #20698: luci-app-acme: "Apply to nginx" does not work.

Curious-r, 2023-03-20:
***

## Steps to reproduce:

1. Install luci-app-acme by offical feed
2. Modify the certificate configuration and check "Apply to nginx"
3. Save and apply

## Actual behavior:

1. Certs was successfully issued in `/etc/acme`, but it was not applied to nginx.

## Expected behavior:

1. Certs apply to nginx rightly

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r0-e467856'
DISTRIB_TARGET='ipq40xx/generic'
DISTRIB_ARCH='arm_cortex-a7_neon-vfpv4'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r0-e467856'
DISTRIB_TAINTS='no-all'
```
The firmware is with luci-ssl-nginx, and all dependencies of luci-app-acme was installed.

Logs:
```
Sun Mar 19 23:15:17 2023 daemon.info acme-acmesh: Running ACME for gateway.suzhou.curious.host
Sun Mar 19 23:15:17 2023 daemon.info acme-acmesh: /usr/lib/acme/client/acme.sh --ecc -d gateway.suzhou.curious.host --keylength ec-384 --accountemail 13582788510@139.com --server letsencrypt --dns dns_cf --issue --home /etc/acme
```

Config of acme:
```
config cert 'curious'
        option enabled '1'
        option use_staging '0'
        option keylength 'ec-384'
        list domains 'gateway.suzhou.curious.host'
        option update_nginx '1'
        option validation_method 'dns'
        option dns 'dns_cf'
        list credentials 'CF_Token="***************************"'
        list credentials 'CF_Account_ID="***************************"'
        list credentials 'CF_Zone_ID="***************************"'
```
Files in `/etc/acme/gateway.suzhou.curious.host_ecc`after certs issued:
```
root@Suzhou:/etc/acme/suzhou.gateway.curious.host_ecc# ls
ca.cer                                suzhou.gateway.curious.host.cer       suzhou.gateway.curious.host.csr.conf
combined.cer                          suzhou.gateway.curious.host.conf      suzhou.gateway.curious.host.key
fullchain.cer                         suzhou.gateway.curious.host.csr
```
uci show nginx:
```
nginx.global=main
nginx.global.uci_enable='true'
nginx._lan=server
nginx._lan.listen='443 ssl default_server' '[::]:443 ssl default_server'
nginx._lan.server_name='_lan'
nginx._lan.include='restrict_locally' 'conf.d/*.locations'
nginx._lan.uci_manage_ssl='self-signed'
nginx._lan.ssl_certificate='/etc/nginx/conf.d/_lan.crt'
nginx._lan.ssl_certificate_key='/etc/nginx/conf.d/_lan.key'
nginx._lan.ssl_session_cache='shared:SSL:32k'
nginx._lan.ssl_session_timeout='64m'
nginx._lan.access_log='off; # logd openwrt'
nginx._redirect2ssl=server
nginx._redirect2ssl.listen='80' '[::]:80'
nginx._redirect2ssl.server_name='_redirect2ssl'
nginx._redirect2ssl.return='302 https://$host$request_uri'
```

Curious-r, 2023-03-20:
> LuCI merely modifies the configuration for the underlying acme package, so the issue likely lies there. @tohojo ?

Thanks! I was about to update this issue. 😆

tohojo, 2023-03-21:
After the refactoring, the support for modifying nginx configuration was removed entirely, on the expectation that it's better if users manually configure their web server to pick up the cert. Acme will still reload the web server (via the hook script in the updated nginx package), but you have to manually configure nginx to use the certificate, which will be symlinked to `/etc/ssl/acme/$MAIN_DOMAIN.{crt,key,fullchain.crt}`

Curious-r, 2023-03-21:
> After the refactoring, the support for modifying nginx configuration was removed entirely, on the expectation that it's better if users manually configure their web server to pick up the cert. Acme [...]

What is the reason to remove the feature? It used to work very well.

tohojo, 2023-03-21:
Separation of concerns, basically: as we added support for more and more daemons, this started to become a bit of a hairball of ad-hoc configuration editing scripts in the acme runner itself. So the refactoring moves to an event-based approach where acme will call hook scripts on renew. The idea being that the packages that want to support auto-configuring themselves can then add their own hook scripts to do this.

At the time of making the change we only added simple reload support to the existing daemons (so that configurations would keep working after an upgrade). And, well, no one has added the auto-config support to any of the packages thus far :)

You mention above that you needed to do some manual re-config of nginx anyway, before this change. What did that consist in, exactly?

tohojo, 2023-03-21:
We never had any code that added a server block directly in acme, so I assume you were using nginx_util for this?

tohojo, 2023-03-21:
Yup, this would be a reasonable thing to add, I think. The way to do it
would be to add a hotplug script *to the nginx_util package* which would
react to new acme events and run the util accordingly...

popy2k14, 2024-02-25:
Hey guys.

I am also setting up nginx on openwrt.
acme.sh is working and i have ticked the renew option in luci acme:

[image]

After issuing (with --force --debug --nginx -> for testing) tzhe following is displayed:

```
[Sun Feb 25 16:13:54 CET 2024] Your cert is in: /etc/acme/....cer
[Sun Feb 25 16:13:54 CET 2024] Your cert key is in: /etc/acme/....key
[Sun Feb 25 16:13:54 CET 2024] The intermediate CA cert is in: /etc/acme/.....cer
[Sun Feb 25 16:13:54 CET 2024] And the full chain certs is there: /etc/acme/....cer
[Sun Feb 25 16:13:54 CET 2024] _on_issue_success
[Sun Feb 25 16:13:54 CET 2024] Run renew hook:'/usr/lib/acme/notify renewed'
[Sun Feb 25 16:13:54 CET 2024] 'dns_duckdns' does not contain 'dns'
[Sun Feb 25 16:13:54 CET 2024] The NOTIFY_HOOK is empty, just return.
```

Is this actually working and nginx reloaded after acme.sh done it's magic?
What makes me curious is the last message:

**[Sun Feb 25 16:13:54 CET 2024] The NOTIFY_HOOK is empty, just return.**

How can i check that nginx is reloaded after acme.sh runned?
Just want to be prepared for the newxt renew.

thx

tohojo, 2024-02-26:
popy2k14 ***@***.***> writes:

> Hey guys.
>
> I am also setting up nginx on openwrt.
> acme.sh is working and i have ticked the renew option in luci acme:
>
> [image]
>
> After issuing (with --force --debug --nginx -> for testing) tzhe [...]

These messages are all coming from acme.sh itself (I'm assuming you also
ran acme.sh itself manually instead of relying on the luci init
script?). The OpenWrt init script sets up its own notification hooks and
calls nginx renew via procd trigger mechanism, but none of that works
when you're running acme.sh manually...

> How can i check that nginx is reloaded after acme.sh runned?
> Just want to be prepared for the newxt renew.

I don't actually know if the service reload is logged anywhere. You
should be able to trigger the ubus message with this command:

ubus call service event '{"type":"acme.renew","data":{}}'

popy2k14, 2024-02-26:
@tohojo thx for your answer 
Yes, I have executed acme.sh from CLI to test it.

So when I got you right, the nginx renew is triggered because of the Cron job (/etc/init.d/acme restart), which runs every night. 

That was I want to know 😊
The missing part was the Openwrt init.d script, which sets up the trigger.

Will see if it works in about 2 months, when the renew is triggered by acme.

Thx

popy2k14, 2024-05-25:
I am using HA now to get my nginxs cert's. So i can not say if the renew worked or not.
But i think it would have worked :-)

So yes, for me it could be closed.