openwrt/packages #20750: meta: Switch "master" branch to "main"

jefferyto, 2023-03-31:
The main repo [voted](https://openwrt.org/voting/2023-02-27) to switch their "master" branch to "main". (The [March 2023 developer meeting notes](https://openwrt.org/meetings/20230328) also has some details.)

I suggest we do the same here, for the simple reason to remain consistent with the main repo.

Not sure if renaming the branch is enough or if we need to do anything more.

aparcar, 2023-04-21:
Other repositories use either the default branch (currently master, then main) or stick to specific branches or commits. To my knowledge a simple rename via the GitHub interface is enough.

aparcar, 2024-03-21:
I'd like to move forward with this and proceed with the rename.

tohojo, 2024-03-21:
No objections from me :)

1715173329, 2024-03-21:
+1. But it would be great to keep master synced up-to-date for compatibility.

BKPepe, 2024-03-21:
Go for it.

nmeyerhans, 2024-03-21:
> +1. But it would be great to keep master synced up-to-date for compatibility.

I think keeping the master branch around (even if it's kept up-to-date) would only lead to confusion. Better to cut ties with the old name entirely.

hnyman, 2024-03-22:
> keeping the master branch around (even if it's kept up-to-date) would only lead to confusion. 

Well, the main OprnWrt has had main and master as an automatic git alias for the last year or two.
No problems so far.

So, I would like main as alias to master, or vice versa.

BKPepe, 2024-10-18:
Can we move forward with this one, @aparcar ? :)

BKPepe, 2025-07-08:
@robimarko Do we have some update regarding this issue, please?