openwrt/packages #20755: lcd4linux-custom: not building all selected drivers

oskarirauta, 2023-04-01:
Maintainer: Florian Eckert / @feckert 
Environment: x86_64, latest git

Description:
When building lcd4linux-custom, it omits driver for digital picture frame (dpf)- even when it's selected.
You can list included drivers of build with `lcd4linux -l` and if you check the build directory, you can find
file drv_dpf.c - without compiled object, dev_dpf.o - so it's excluded from build.

On line 219 of package's Makefile, configure args for lcd4linux-custom build, are defined. If you change it to look like this:
```
  CONFIGURE_ARGS+= \
        --with-drivers="$(LCD4LINUX_CUSTOM_DRIVERS) DPF" \
        --with-plugins="$(LCD4LINUX_CUSTOM_PLUGINS)" \
```

you force inject driver dpf to list of drivers to be built, and it gets built - so there must be something wrong with the Makefile. I can not verify if this happened to any other driver as well; but at least most of them are built-in, I just noticed that dpf was excluded.

feckert, 2023-05-23:
Fixed with commit https://github.com/openwrt/packages/commit/05b0c5be88cf9e407db451f2cca8d8a257bcc602
The name was wrong! The module is not called `DPF` and not `dpf`

feckert, 2023-05-23:
@oskarirauta Thanks for the hint

closed 2023-05-23 by feckert

oskarirauta, 2023-05-23:
@feckert - No. Thanks to you for fixing it :)