openwrt/packages #20756: rust: wrong ABI on PPC64 with musl

stintel, 2023-04-01:
Maintainer: @lu-zero 
Environment: OpenWrt master r22463-3c3d797c4d qoriq/generic

Description:

Building rust packages (e.g. ripgrep) for the qoriq target fails in the linker phase:

```
/home/stijn/Development/OpenWrt/openwrt/staging_dir/toolchain-powerpc64_e5500_gcc-12.2.0_musl/lib/gcc/powerpc64-openwrt-linux-musl/12.2.0/../../../../powerpc64-openwrt-linux-musl/bin/ld: /home/stijn/Development/OpenWrt/openwrt/build_dir/target-powerpc64_e5500_musl/ripgrep-13.0.0/target/powerpc64-unknown-linux-musl/stripped/deps/rg-80c10b6350b96ea9.rg.5e4cd94a-cgu.0.rcgu.o: ABI version 1 is not compatible with ABI version 2 output
```

This is because musl always uses the ELFv2 ABI on PPC64, regardless of endianness. This was fixed for toolchain/gcc in https://github.com/openwrt/openwrt/commit/6e2556963d523ecef9a2a6130c899d146a3b7333, but I'm not immediately finding how to do this for rust.

lu-zero, 2023-04-01:
Can you direct me to reproduce the problem locally?

stintel, 2023-04-01:
Just select qoriq target, enable ripgrep, and try to build.

lu-zero, 2023-04-01:
It is also nominally fixed in rustc:
``` rust
    let abi: ABI = if cx.target_spec().env == "musl" {
        ELFv2
    } else {
        match cx.data_layout().endian {
            Endian::Big => ELFv1,
            Endian::Little => ELFv2,
        }
    };
```

stintel, 2023-04-01:
I've seen that, but it doesn't seem to work. I've been digging for a few hours already, but no clue.

lu-zero, 2023-04-01:
There isn't a ready to use snapshot?

stintel, 2023-04-01:
> There isn't a ready to use snapshot?

Snapshot image? The target is source-only...

lu-zero, 2023-04-01:
Then I'll need a bit of time to get to that, I wonder why it survived in the CI...

stintel, 2023-04-01:
CI doesn't build ppc64.

stintel, 2023-06-29:
@lu-zero any ideas?

lu-zero, 2023-06-29:
I need to find time to test locally, if you can point me a .config so I'm sure we are on the same page better. Ping me this weekend if I do not reply before :)

stintel, 2023-07-27:
This appears to be fixed!

closed 2023-07-27 by stintel