openwrt/packages #20761: unbound-control: Looking for SSL Cert in the wrong directory

zweistein2010, 2023-04-02:
Maintainer: @BKPepe @EricLuehrsen 
Environment: (put here arch, model, OpenWrt version) ARMv7, GL.iNet GL-B1300, Version 22.03

Description:

The following packages were installed: luci-app-unbound, unbound-control, unbound-control-setup

During setup I wanted to enable the following in luci:
Services->Recursive DNS->Unbound->Resource
Unbound Control App: Local Host, Encrypted

The certificates are generated in /etc/unbound but unbound-control seems to expect them in /var/lib/unbound/etc/unbound/unbound_control.pem

When trying to query the unbound-conrol status via terminal I will get the following error
`root@OpenWrt:~# unbound-control status
error: Error setting up SSL_CTX client cert
/var/lib/unbound/etc/unbound/unbound_control.pem: No such file or directory
root@OpenWrt:~# `
Manually creating the expected directory and adding the cert fixes the problem until the devices is restarted. I was unable to find an option to point unbound-control to the correct location.

It would be great if this could be fixed! Thank you

zweistein2010, 2023-04-02:
After some further research I found this to be related with the following post:
https://forum.openwrt.org/t/need-help-with-unbound-setup-on-snapshots/86245

Setting unbound to 'Local Host, unencrypted' circumvents the certificate problem and the unbound crash on start-up. But it would be great to actually be able to use unbound-control with an encrypted connection.

EricLuehrsen, 2023-04-25:
When unbound drops from privileged to user mode, it chroot's and hides the file system from itself. The configuration file path is virtually shifted. This works **within** unbound PID and the /var/... config file reflects that relativity. Calling unbound-control from the command line is not chroot'd, so it gets confused. You can create an alias (literal shell alias) and pass unbound-control command line options for pem files. This is how the UCI and LuCI scripts work.