openwrt/packages #20772: <bluez> [BUG] A use after free bug in /src/gatt-database.c

ShangzhiXu, 2023-04-04:
Maintainer: /
Environment: 22.3

Description:

In /src/gatt-database.c   [Line 3878] called
```
send_notification_to_device(state, state->pending);
```
In this function, 
```
static void send_notification_to_device(void *data, void *user_data)
       struct device_state *device_state = data;
	struct notify *notify = user_data;
...
...
if (!server) {
	if (!device_is_bonded(device, device_state->bdaddr_type))
			goto remove;
	state_set_pending(device_state, notify);
	return;
}
...
...
remove:
	/* Remove device state if device no longer exists or is not paired */
	if (queue_remove(notify->database->device_states, device_state)) {
		queue_foreach(device_state->ccc_states, clear_ccc_state,
						notify->database);
		device_state_free(device_state);
	}

```
In the `remove` label, `device_state` will get freed in function `device_state_free()`; which means the `state` in the first section will be freed.
device_state_free:
```
static void device_state_free(void *data)
{
	struct device_state *state = data;

	queue_destroy(state->ccc_states, free);

	if (state->pending) {
		free(state->pending->value);
		free(state->pending);
	}

	free(state);
}
```

But after it has been freed,  in [Line 3878], `state` is used again.
Which might lead to a UAF bug.

**Suggestion:**
I think it can be solved by adding a patch to check if session has been freed. By the way this bug has been fixed on the newest version of libcoap on github.

karlp, 2023-04-07:
so, just submit a PR to update bluez?  this really isn't the right place to report issues in upstream projects...

diizzyy, 2023-07-30:
Closing, not a packaging issue and should be submitted upstream

closed 2023-07-30 by diizzyy