openwrt/packages #20775: php-cli: 8.1.16 Segmentation fault with date();

hoeulm, 2023-04-04:
OpenWrt package php-cli.8.1.16

this packages crashes  with "Segmentation fault" on the simple program

ymdhms.php
<?php echo(date("Y-m-d H:i:s")); ?>

OpenWrt 22.03.3
Board   ramips/mt7621 on DLINK  882 A1

other more complex programs with sockets and openssl-decrypt work FINE

On an other OpenWrt with similar CPU the program runs without problems!!
Board   ramips/mt7620 on GL-MT300A

mhei, 2023-04-04:
Could you provide a link to the script which crashs PHP?
Do you build on your own, or do you use precompiled packages?
When compiling on your own, then please provide output of "scripts/diffconfig.sh".
What packages are installed? Have both systems the same package set?
Which timezone packages are installed?

mhei, 2023-04-11:
I can only reproduce a Segmentation fault when I do the following:
`pkg remove zoneinfo-core --force-depends`
Can you check whether the corresponding zoneinfo-* package is installed, or if you don't set a timezone, then you need at least the `zoneinfo-core` package.

hoeulm, 2023-04-11:
Hello Michael,

there was NO zoneinfo Package installed, if i install it, all works 
fine!

It was as default installation of openwrt,

so "zoneinfo-core" should be added to the depencies of php-cli

Thanks for your help!!

I develop a Smart-Energy Monitor Interface with  REVOGI SOW323 / 
VOLTCRAFT SEM8500 Powerstripes that runs on OpenWrt instead of IOBROKER 
or APPS,

so you dont need to install APPS, only use your Browser(also on 
Laptops!)

Richard Hoermann

Links:
------
[1] 
https://github.com/openwrt/packages/issues/20775#issuecomment-1504020129
[2] 
https://github.com/

mhei, 2023-04-15:
Hi,
thanks for clarifying. Then this is expected behavior - not nice, but I don't see an easy solution to fix this: Luci allows to select the timezone independently from what is installed as zoneinfo-* packages as far as I see. Maybe you want to open an issue for Luci?
I tend to close this issue here if you agree.

mhei, 2023-04-17:
Not a bug in PHP, closing as announced.

closed 2023-04-17 by mhei