openwrt/packages #20782: Plugin 'wolfssl' failed to load: Error relocating /usr/lib/ipsec/plugins/libstrongswan-wolfssl.so: wolfssl_ec_diffie_hellman_create: symbol not found

vochong, 2023-04-07:
@stintel

When strongswan ipsec or swanctl is started,  this error is seen:

plugin 'wolfssl' failed to load: Error relocating /usr/lib/ipsec/plugins/libstrongswan-wolfssl.so: wolfssl_ec_diffie_hellman_create: symbol not found

# ls -al /usr/lib/ipsec/plugins/libstrongswan-wolfssl.so
-rwxr-xr-x    1 root     root         34873 Apr  6 03:21 /usr/lib/ipsec/plugins/libstrongswan-wolfssl.so

# cat /etc/os-release 
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r22506-9a62b3977f"
OPENWRT_BOARD="ipq806x/generic"
OPENWRT_ARCH="arm_cortex-a15_neon-vfpv4"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r22506-9a62b3977f"

stintel, 2023-04-07:
I'm not maintaining strongswan anymore.

neheb, 2023-04-14:
@cotequeiroz should know

cotequeiroz, 2023-04-18:
I suspect phase1 and phase2 bots are producing wolfssl libraries with different options. Is strongswan or any other feed package forcing any WOLFSSL_* (DEPENDS+=@WOLFSSL_*) option?  I'm away from a computer now, and it's hard to even navigate through code with a phone.