openwrt/packages #20793: isc-dhcp is end of life - remove

rbeede, 2023-04-11:
### Describe the bug

ISC DHCP is end of life and no longer receiving bug fixes. It should be removed as a package from the next major release.

From https://www.isc.org/dhcp/ (taken 4/11/2023):
>ISC has announced the end of maintenance for ISC DHCP as of the end of 2022.

OpenWrt already has other alternatives for DHCP clients and servers:

- odhcpd (including a DHCPv4 special option)
- dnsmasq
- dhcpcd
- 

### OpenWrt version

r20028-43d71ad93e

### OpenWrt target/subtarget

x86/64

### Device

LENOVO 10HY0021US

### Image kind

Official downloaded image

### Steps to reproduce

opkg list | grep isc-dhcp

### Expected behaviour

Removal of the old  isc-dhcp  packages

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

pprindeville, 2023-04-11:
ISC-DHCP is still receiving security fixes, as evinces by my recent update to 4.4.3-P1.

As a replacement, I'll look at Kea (also from ISC).

brada4, 2023-04-12:
The remote dns update works (over localhost), though migration tool spits note that may not get migrated, so small manual integration was needed.
Maybe package that migration tool and slowly fade off client and relay that ISC claim to never update again?

pprindeville, 2023-07-26:
I'm still working on porting over the wrappers that parse `/etc/config/dhcp` and generate the `dhcpd.conf` file to generating KEA output instead.

When that's done, and merge, we can wait 90 days and then delete the project.

pprindeville, 2023-11-05:
Update: I've got a few PR's for `openwrt` base and `libubox` that need to be merged to accommodate the migration to Kea that are stalled:

https://github.com/openwrt/libubox/pull/6
https://github.com/openwrt/openwrt/pull/13765

If anyone wants to review them, that might help things along...

Neustradamus, 2025-11-10:
The migration to Kea is important.