openwrt/packages #20800: ddns-scripts: include samples inside the package

perrin4869, 2023-04-13:
# Issue template (remove lines from top till here)

Maintainer: @feckert 
Environment: OpenWrt 22.03.3 r20028-43d71ad93e

Description:

I am interested in using the `slaac_sample.sh` inside my own OpenWrt box, but interestingly it's just not included in the distributed package. I think it would be useful to include it, since if there are updates, we don't need to keep track of the file inside the git repo tree, and because the file is useful as-is.

feckert, 2023-04-18:
I must confess I have never used the script before.
What do you need it for?
Can the information be fetched from the ubus?

perrin4869, 2023-04-18:
this is useful for setting a ddns record to the value of a down stream ipv6 node, in the case where the node is using slaac. Since this value is generated dynamically by the downstream node, I don't think it can be fetched by `ubus`...

perrin4869, 2023-08-23:
@feckert thought I'd ping you about this :)
I've been using that script inside my OpenWRT box (just copied it over to my `/etc/config` directory) for a few months, and haven't had any issues, it works perfectly
Something extra that I noticed, those sample scripts [seem to be expecting to be available in `/usr/lib/ddns`](https://github.com/openwrt/packages/blob/f3a6491674ab451535e392b63e94a545abf3f292/net/ddns-scripts/samples/slaac_sample.sh#L9). The same kind of comment can be found in [`getlocalip_sample.sh`](https://github.com/openwrt/packages/blob/f3a6491674ab451535e392b63e94a545abf3f292/net/ddns-scripts/samples/getlocalip_sample.sh#L9) and [`update_sample.sh`](https://github.com/openwrt/packages/blob/f3a6491674ab451535e392b63e94a545abf3f292/net/ddns-scripts/samples/update_sample.sh#L6). I am not familiar with the build tooling for openwrt packages, I tried to have a look at the makefile, but there's a lot of magic going on there ^^;;
I can try to put together a PR some weekend when I have time, but if you could do that, I'd be eternally grateful!
Should note that they need to have `+x` permissions to be useful.

feckert, 2023-10-04:
Sorry I don't use the script and I don't have time for packaging this. If you want to have it, then you can create a pullrequest. Please have a look at other Makefiles in in this repository to understand how the packaging works in OpenWrt. I could review your pullrequest.

closed 2023-10-04 by feckert

perrin4869, 2023-10-04:
no worries :)
the Makefiles, last time I looked at them, looked rather complicated, but I'll give it a try sometime in the future

PR #22303 (2023-10-09): ddns-scripts: add samples to package
Maintainer: @feckert

Description: closes #20800
I am not sure how to run this `Makefile`. I tried to drop it inside the `openwrt` root tree, then run `make`, but it didn't work, so couldn't test how this works...