openwrt/packages #20818: htop package not available

iluvatyr, 2023-04-16:
Maintainer: [@\<github-user> (find it by checking history of the package Makefile)](https://github.com/champtar)
Environment:

Model | ASUS RT-AX53U
Architecture | MediaTek MT7621 ver:1 eco:3
Target Platform | ramips/mt7621
Firmware Version | OpenWrt 22.03.3 r20028-43d71ad93e / LuCI openwrt-22.03 branch git-22.361.69894-438c598
Kernel Version | 5.10.161

Description:
I get error when installing package "htop"
https://downloads.openwrt.org/releases/22.03.3/packages/mipsel_24kc/packages/htop_3.2.1-1_mipsel_24kc.ipk 
It can be seen that there is a nginx error 404.

```

root@OpenWrt:~# opkg update && opkg install htop
Installing htop (3.2.1-1) to root...
Downloading https://downloads.openwrt.org/releases/22.03.3/packages/mipsel_24kc/packages/htop_3.2.1-1_mipsel_24kc.ipk
Collected errors:
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.3/packages/mipsel_24kc/packages/htop_3.2.1-1_mipsel_24kc.ipk, wget returned 8.
 * opkg_install_pkg: Failed to download htop. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package htop.
```

brada4, 2023-04-16:
Just to confirm package is absent for 22.03.4 as well.

champtar, 2023-04-17:
I see
- https://downloads.openwrt.org/releases/22.03.3/packages/mipsel_24kc/packages/htop_3.2.2-1_mipsel_24kc.ipk
- https://downloads.openwrt.org/releases/22.03.4/packages/mipsel_24kc/packages/htop_3.2.2-1_mipsel_24kc.ipk

and the index (`Packages.gz`) also points to 3.2.2, so maybe you tried to update while the package was updating ? or the CDN as some stale data ?

Can you retry ?

iluvatyr, 2023-04-17:
Hello,
I double checked before creating the issue with several other users who also had the problem through the normal update process.
The issue is resolved now though, it seems. Can be closed

brada4, 2023-04-17:
Use the buttons below chat box to do so....

closed 2023-04-17 by iluvatyr