openwrt/packages #20841: <php7> php7-mod-* package not load

brett90063, 2023-04-19:
Maintainer: @mhei 
Environment: 
 - OpenWrt: 21.03
 - PHP: 7.4.24 (cli) ( NTS )

Description:

Hello,
I got `Fatal error: Uncaught Error: Call to undefined function openssl_decrypt()` message in my lighttpd using php file.

I already have `php7-mod-openssl`, but `php-cli -m` is not showing openssl. 
When I add another package like `php7-mod-phar`, `php-cli -m` also doesn't show phar. 
I have installed about eight different mods for testing purposes, but they don't work. :(

Here are my package and config:
1. `opkg list`:
```
# opkg list |grep php
php7 - 7.4.24-1
php7-cgi - 7.4.24-1
php7-cli - 7.4.24-1
php7-fastcgi - 7.4.24-1
php7-fpm - 7.4.24-1
php7-mod-openssl - 7.4.24-1
php7-mod-phar - 7.4.24-1
```
2. `php-cli -m`:
```
# php-cli -m
[PHP Modules]
Core
date
hash
pcre
posix
Reflection
SPL
standard
zlib

[Zend Modules]
```
3. `php-cli --ini`:
```
# php-cli --ini
Configuration File (php.ini) Path: /etc
Loaded Configuration File:         /etc/php.ini
Scan for additional .ini files in: /etc/php7
Additional .ini files parsed:      /etc/php7/15_openssl.ini,
/etc/php7/20_phar.ini
```
```
# cat /etc/php7/15_openssl.ini /etc/php7/20_phar.ini 
extension=openssl.so
extension=phar.so
```
4. .so file:
```
# cat /etc/php.ini |grep extension
extension_dir = "/usr/lib/php"
```
```
# ls /usr/lib/php/
openssl.so  phar.so
```
Looks like some reason cause mod can't load.
Am I missing something? any suggestion will be appreciated, thanks in advance.

brada4, 2023-04-19:
```
php7-mod-openssl - 7.4.33-1
root@OpenWrt:/# php-cli -m
[PHP Modules]
Core
date
hash
libxml
openssl
```

mhei, 2023-04-21:
Just to ensure: `opkg list | grep openssl` reports what?

brett90063, 2023-04-21:
@mhei 
```
# opkg list |grep openssl
libopenssl-conf - 1.1.1t-1
libopenssl1.1 - 1.1.1t-1
libustream-openssl20201210 - 2020-12-10-68d09243-1
openssl-util - 1.1.1t-1
openvpn-openssl - 2.5.3-3
php7-mod-openssl - 7.4.24-1

```

mhei, 2023-04-22:
In your first post, you mentioned php version 7.4.24-1, then 7.4.33-1 and finally again 7.4.24-1. I hope that you have not mixed different versions in you installation?
With the latest 21.02 branch, self-compiled - I cannot reproduce on my favorite ARM based system.
Do you compile yourself, or use pre-compiled packages?

brada4, 2023-04-22:
I am not OP, I mentioned that fully upgraded php loads modules successfully on a fully upgraded 21.02.

mhei, 2023-04-22:
Indeed, sorry for my confusion. :innocent:  Maybe OP has a chance to fully upgrade and re-try then.

brett90063, 2023-04-24:
Hi all,
Sorry for late reply. After upgrading php from version 7.4.24-1 to 7.4.33-1, `php-cli -m` still not show openssl.

> Do you compile yourself, or use pre-compiled packages?

I am not certain of what I used, but I utilized menuconfig and compiled it. Like upgrade the package, I only need to modify the packages/lang/php7/Makefile.

That's weird, maybe some of package will conflick with php-extension? because all of php7-mod is invalid in my enviorment.

mhei, 2023-04-24:
Using menuconfig and compile sounds like self-compiling :smile:
But for upgrading the php package it should not be necessary to edit the Makefile yourself, just use git to upgrade to the latest branch heads.
Maybe you could try to run
```
scripts/feeds update -a && scripts/feeds install -a && make oldconfig && make
``` 
and check which package versions are then included in your image and whether this changes anything for you?

brett90063, 2023-04-25:
Hi @mhei 
I can't use `scripts/feeds` to update, install and make, because it's control by docker
it shows:
```
Makefile:16: *** Please use docker container to compile. Use scripts under tools/.  Stop.
```
This is provide by other department in my company, so im not good at this.

But anyway...I can and already update php to 7.4.33-1
```
# opkg list |grep php
php7 - 7.4.33-1
php7-cgi - 7.4.33-1
php7-cli - 7.4.33-1
php7-fastcgi - 7.4.33-1
php7-fpm - 7.4.33-1
php7-mod-ftp - 7.4.33-1
php7-mod-imap - 7.4.33-1
php7-mod-openssl - 7.4.33-1
php7-mod-snmp - 7.4.33-1
```
```
#opkg list |grep openssl
libopenssl-conf - 1.1.1t-1
libopenssl1.1 - 1.1.1t-1
libustream-openssl20201210 - 2020-12-10-68d09243-1
openssl-util - 1.1.1t-1
openvpn-openssl - 2.5.3-3
php7-mod-openssl - 7.4.33-1
```
but same problem :(
also .ini and .so etc. config setup is same with first comment
```
# php-cli -m
[PHP Modules]
Core
date
hash
pcre
posix
Reflection
SPL
standard
zlib

[Zend Modules]

```

brada4, 2023-04-25:
Standard php module works, so you have to ask middleman to give it to you.

mhei, 2023-04-25:
@brett90063 : please check company internal about the whole process and how to upgrade. If you don't have full control over the build process and you cannot describe exactly what you do different to an "usual" build process, then it is very hard to help you.

brett90063, 2023-04-26:
OK, sorry for my lack of knowledge about the company's internal process, I'll check with the relevant team and gather more information to better describe the differences in my build process.
@brada4 @mhei Thanks for both your time, really appreciate it.
Now I close it.

closed 2023-04-26 by brett90063