openwrt/packages #20848: strongswan: "running with no instances"

lowjoel, 2023-04-20:
Maintainer: @pprindeville 
Environment: aarch64 Linksys E8450, OpenWrt 22.03.4

Description:

I've noticed that when I specify `ipsec.interface` on `/etc/config/ipsec`, occasionally Charon doesn't actually listen for connections. `/etc/init.d/swanctl status` says `running with no instances`.

I tried to dig through the code and my main suspect is: https://github.com/openwrt/packages/blob/8686e69765d2e2886c7819eacb55bbd7bebe170e/net/strongswan/files/swanctl.init#L715

I believe that's a problem when swanctl-init runs before the interfaces are actually registered (maybe a race with interface bringup?)

I'm not sure where this problem originated, but maybe the fix is to provide a default /etc/hotplug.d/iface script that reloads the service every time an interface change is detected? If that seems to be the fix I'll be happy to provide a patch.

aleks-mariusz, 2023-09-30:
@lowjoel i'd be curious to see what this patch looks like.. 

at the moment i have put this in my rc.local as a hacky workaround:

```
(  /etc/init.d/swanctl stop; sleep 10; /etc/init.d/swanctl start ) &
```

pprindeville, 2023-10-23:
> I'm not sure where this problem originated, but maybe the fix is to provide a default /etc/hotplug.d/iface script that reloads the service every time an interface change is detected? If that seems to be the fix I'll be happy to provide a patch.

Alternately, we could add an interface trigger.  Maybe something like:

```
service_triggers() {
        procd_add_reload_interface_trigger wan

        ...
}
```

lowjoel, 2023-10-23:
@aleks-mariusz yeah, that in a hotplug.d/iface script would probably be more predictable. But that's essentially what @pprindeville's patch using `prod_add_reload` does, which would be the better fix. Maybe in /etc/init.d/swanctl:

```bash
service_triggers() {
    config_foreach service_trigger_ipsec ipsec

    ...
}

service_trigger_ipsec() {
    local interface
    local interface_list
    config_list_foreach "$1" interface append_var interface_list
    for interface in $interface_list; do
        procd_add_reload_interface_trigger $interface
    done
}
```

(hacked together with zero testing, maybe you can think of a better solution)

pprindeville, 2023-11-04:
> @aleks-mariusz yeah, that in a hotplug.d/iface script would probably be more predictable. But that's essentially what @pprindeville's patch using `prod_add_reload` does, which would be the better fix. Maybe in /etc/init.d/swanctl:
> 
> ```shell
> service_triggers() {
>     config_foreach service_trigger_ipsec ipsec
> 
>     ...
> }
> 
> service_trigger_ipsec() {
>     local interface
>     local interface_list
>     config_list_foreach "$1" interface append_var interface_list
>     for interface in $interface_list; do
>         procd_add_reload_interface_trigger $interface
>     done
> }
> ```
> 
> (hacked together with zero testing, maybe you can think of a better solution)

I'm not sure there's any value to adding triggers to LAN interfaces...  Though I can see the need for supporting multiple WAN devices if someone is multi-homed and using `mwan3`, for instance.

Please provide a tested patch and I'll review it.

Also, I'd put as many variables on a single `local` line as will fit for compactness.

lowjoel, 2023-11-13:
I've applied that [patch](https://github.com/openwrt/packages/issues/20848#issuecomment-1775379577) in /etc/init.d/swanctl after upgrading my routers to 5.9.11 -- it's hard to verify, but it does feel like the likelihood of "running with no instances" seem to have dropped drastically. I am indeed using `mwan3`.

I can combine the locals in the same line, but the question is -- even if there's no interface specified, we should be reloading the config every time a network interface changes? In my case:

```
 config ipsec
        list interface 'wan'
        list interface 'wanb'
```
Makes sense, but if no interfaces are specified, no changing config (even with mwan3) would cause a reload. Any ideas?

PR #22898 (2023-12-24): strongswan: trigger reload when interfaces are specified
Maintainer: @pprindeville @Thermi 
Compile tested: na
Run tested: aarch64/mt7622 Linksys E8450

Description:

Add interface triggers if interfaces to listen to are specified in `/etc/config/ipsec`. This fixes the "running with no instances" scenario after rebooting a router.

I've been running this for a month to good effect.

Fixes #20848
commit f2d209e4 (2023-12-24): strongswan: trigger reload when interfaces are specified

Fixes #20848

Add interface triggers if interfaces to listen to are specified in
`/etc/config/ipsec`. This fixes the "running with no instances" scenario
after rebooting a router.
commit 4e1c175d (2024-01-02): strongswan: trigger reload when interfaces are specified

Fixes #20848

Add interface triggers if interfaces to listen to are specified in
`/etc/config/ipsec`. This fixes the "running with no instances" scenario
after rebooting a router.

(cherry picked from commit f2d209e4fff3048883ee5953c3a312d9f5aaab1b)
closed 2023-12-24 by pprindeville